daisyaudio/libDaisy #188: Provide metadata file (e.g. JSON) to document a specific hardware's IO features

grrrwaaa, 2020-05-30:
There are already 4 hardware configurations, and doubtless there will be more as users embed the  Seed into other circuits. For the code generation it would be very helpful to be able to know what inputs and outputs are available, including what their expectations are (e.g. voltage ranges, sampling rates/frequency ranges, DC/AC coupled, etc.) in order to generate code (or user error feedback) accordingly. 

A JSON file hardware descriptor would be great for that.

stephenhensley, 2020-05-30:
here was what I originally had started building (before starting work on any generators). So there's definitely more information that can be provided, but I'm definitely open to ideas of what should/shouldn't be in there.

Also, for certain things (like RGB LEDs) it almost seems like they should be provided to generators as 3 single LEDs for the sake of common access. Not sure what would be most user friendly in pd or gen~ there: `[send @comment led1 0.2 0.5 1.0]` or `[send @comment led1_red 0.2]`

I can see wanting to include voltages and such for user feedback, but libdaisy has pretty much all of this stuff normalized to 0-1.

I'm not sure exactly how we want to handle encoders. The simplest would be having it act as 'bangs' for increments, and basically a button for the click-segment. 

Here are the controls for the pod.

```
{
  "knob1": {
    "type": "AnalogControl",
    "descriptor": "POT",
    "io": "INPUT"
  },
  "knob2": {
    "type": "AnalogControl",
    "descriptor": "POT",
    "io": "INPUT"
  },
  "encoder": {
    "type": "Encoder",
    "descriptor": "ENCODER",
    "io": "INPUT"
  },
  "button1": {
    "type": "Switch",
    "descriptor": "BUTTON",
    "io": "INPUT",
  },
  "button2": {
    "type": "Switch",
    "descriptor": "BUTTON",
    "io": "INPUT",
  },
  "led1": {
    "type": "RgbLed",
    "descriptor": "RGBLED",
    "io": "OUTPUT"
  },
  "led2": {
    "type": "RgbLed",
    "descriptor": "RGBLED",
    "io": "OUTPUT"
  },
}
```

andrewikenberry, 2020-10-28:
[image] [Create Global Metadata format for breakout boards](https://trello.com/c/TEUQ04Jw/5-create-global-metadata-format-for-breakout-boards)

stephenhensley, 2021-09-03:
Just wanted to update this since it's been around for so long without a lot of activity.

We have made some progress, and have the basic workings set up for pd2dsy, and oopsy.

Long-term plan being that the board support c++ files can be generated from metadata rather than have to be maintained by hand.

This helps with two key things:

1. It allows us to manage many more board files within the libDaisy ecosystem without the need to manually manage them when API changes occur. It is also an easier way for integrating the Daisy hardware into other code generators, etc.
2. It makes the creation of new hardware based on the Daisy trivial, and removes the requirement to write any c++ to get started with pd or max when using custom hardware.

Here is an example of what the meta data looks like for the Daisy Pod:

```json
{
	"name":"pod",
	"defines": {
		"OOPSY_TARGET_HAS_MIDI_INPUT": 1
	},
	"display": {
		"config": {
			"thing1": "conf1;",
			"thing2": "conf2;",
			"thing3": "conf3;"
		}
	},
	"components": {
		"sw1": {
			"init": "test",
			"component": "Switch",
			"pin": 27
		},
		"sw2": {
			"component": "Switch",
			"pin": 28
		},
		"knob1": {
			"component": "AnalogControl",
			"pin": 21
		},
		"knob2": {
			"init": "test",
			"component": "AnalogControl",
			"pin": 15
		},
		"encoder": {
			"component": "Encoder",
			"pin": {"a":26, "b":25, "click":13 }
		},
		"led1": {
			"component": "RgbLed",
			"pin": {"r":20, "g":19, "b":18 }
		},
		"led2": {
			"component": "RgbLed",
			"pin": {"r":17, "g":24, "b":23 }
		},
		"gateout": {
			"component": "GateOut",
			"pin": 1998,
			"display": "display code for gateout;"
		},
		"led3": {
			"component": "Led",
			"pin": 100,
			"display": "display code for led3;"
		},
		"cvout": {
			"component": "CVOuts"
		},
		"gatein": {
			"component": "GateIn",
			"pin": 12
		},
		"sw3": {
			"component": "Switch3",
			"pin": {"a": 1, "b": 2}
		}

	},
	"aliases": {
		"switch": "sw1",
		"button": "sw1",
		"switch1": "sw1",
		"button1": "sw1",
		"switch2": "sw2",
		"button2": "sw2",
		"enp": "encoder_press",
		"switch3": "encoder_press",
		"press": "encoder_press",
		"knob": "knob1",
		"ctrl": "knob1",
		"ctrl1": "knob1",
		"ctrl2": "knob2",
		"led": "led1"
	}
}
```

This can be used to accurately recreate the board support files for the Daisy Pod. 

There is node js code for doing the conversion built into oopsy, and there is a separate python module that's been added to do so within pd2dsy.

Support for things like displays, multiplexors, shift registers, etc. are still not accessible.