daisyaudio/libDaisy #194: Add MIDI enumeration and event support for USB OTG host

ggsubs, 2020-06-04:
Current codebase does not support USB Midi devices to be plugged into Daisy's OTG host. It is the feature request to add support and integrate into the midi event handler.

polyclash, 2020-08-13:
Any eta in this?

polyclash, 2020-08-13:
Any eta in this?

polyclash, 2020-08-13:
Any eta in this?

polyclash, 2020-08-13:
Any eta in this?

polyclash, 2020-08-13:
Any eta in this?

polyclash, 2020-08-13:
Any eta in this?

polyclash, 2020-08-13:
Any eta in this?

polyclash, 2020-08-13:
Any eta in this?

polyclash, 2020-08-13:
Any eta in this?

stephenhensley, 2020-08-13:
There are few things related to the MIDI parser, and adding USB MIDI Device support that are next up in the priority list before beginning to add USB Host support. 

Hoping to address those in the next week or two, and then we'll be able to put some time into this.

polyclash, 2020-08-13:
Good.
The midi parse will support the sysex message?