daisyaudio/libDaisy #195: Dfu programming error when multiple devices detected

liquidcitymotors, 2020-06-05:
Hi!

Thank you for this inspiring project.

Compilation was very smooth, but I ran into a minor issue when programming over USB.

`make program-dfu` exited before programming with the following error:

```console
dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device
```

`dfu-util -l` showed this:

```console
Found Runtime: [05ac:8290] ver=0167, devnum=4, cfg=1, intf=5, path="20-8", alt=0, name="UNKNOWN", serial="UNKNOWN"
Found DFU: [0483:df11] ver=0200, devnum=11, cfg=1, intf=0, path="20-5", alt=1, name="@Option Bytes   /0x5200201C/01*128 e", serial="200364500000"
Found DFU: [0483:df11] ver=0200, devnum=11, cfg=1, intf=0, path="20-5", alt=0, name="@Internal Flash   /0x08000000/16*128Kg", serial="200364500000"
```

That "Runtime: [05ac:8290]" device is actually present on my machine (2015 Macbook Pro, 10.14) even with no USB devices connected. I did some sleuthing, and it looks to be the computer's internal bluetooth/wifi peripheral.

It's hard to suggest steps for reproduction because that device is definitely the culprit. I figured if anything, it would be good to open an issue to see if anyone else has the same problem.

I noticed that the `program-dfu` rule in the Makefile does not use the `--device` option. Adding that option as `--device 0483:df11,0483:df11` resolved the issue for me, but I'm not confident hard coding the chipset is the right move. `--device ,` might be a more general solution, as it will only be tripped up when another device in DFU mode is detected.

Hopefully this illustrates the issue sufficiently, but let me know if you would like any more information. Now the fun begins!

stephenhensley, 2020-06-05:
Noted! I'll look into adding the `--device` flag this weekend. Since the system bootloader has its own vid/pid pair I think that hard coding it should be okay, but I'll do some double checking.

stephenhensley, 2020-07-13:
We've added the `-d` flag for the specific vid/pid combo for the Daisy Seed's Bootloader which fixes this.

closed 2020-07-13 by stephenhensley