daisyaudio/libDaisy #197: UART1 and Midi not working on Seed board

ggsubs, 2020-06-07:
The midi input is not working when building on Mac, with gcc-arm-none-eabi-9-2019-q4-major. The code is based on PodMidi. Midi is sent in on pin 37 of the Seed board. The TX pin of UART1 (36 on Seed) is not pulled high, assuming the UART1 hardware is not initialized. I excluded hardware fault by blinking a led on D20/D30 pins using, STM32H7 Arduino with the Daisy variant. Neither Arduino Serial working on UART1.

ggsubs, 2020-06-07:
Closing - the issue is with the Daisy Pinout. USART1 is indicated for both pins 36/37 and 14/15. The correct Seed pins are 14/15.

closed 2020-06-07 by ggsubs

stephenhensley, 2020-06-08:
@ggsubs sorry for the confusion.

The chip itself is capable of changing which pins use which peripherals. However, each peripheral is hard-coded to a single set of pins in the Arduino (and temporarily in the UART driver in libdaisy as all of the configuration has yet to be added).