daisyaudio/libDaisy #202: UI Event Queue

TheSlowGrowth, 2020-06-18:
This relates to #201 

To help users design their software more cleanly, it would be helpful to have a UI event queue that can queue user input events like

1. button pressed
2. button released
3. encoder turned
4. potentiometer turned
5. SD card added or removed (?)

Let's discuss what features would be required and how this could be integrated into the daisy framework.

TheSlowGrowth, 2020-06-18:
Here's a suggestion for a simple implementation. I post this here as a starting point for discussing the required features.
````
/** A queue that holds user interface events such as button presses or encoder turns.
  */
class UiEventQueue
{
	/** An event */
	struct Event __attribute__((packed))
	{
		/** The type of event */
		enum class EventType
		{
			invalid,
			buttonPressed,
			buttonReleased,
			encoderTurned,
			potMoved
		};

		/** The type of event that this Event object represents. */
		EventType type : 16; // default size could be more than 16 bits - not required

		union {
			struct {
				/** The unique ID of the button that was pressed. */
				uint16_t id;
			} asButtonPressed;
			struct {
				/** The unique ID of the button that was released. */
				uint16_t id;
			} asButtonReleased;
			struct {
				/** The unique ID of the encoder that was turned. */
				uint16_t id;
				/** The number of increments detected. */
				int16_t increments;
				/** The total number of increments per revolution. */
				uint16_t stepsPerRev;
			} asEncoderTurned;
			struct {
				/** The unique ID of the pot that was moved. */
				uint16_t id;
				/** The new position of the pot. */
				float newPosition;
			} asPotMoved;
		};
	};

	UiEventQueue() {}
	~UiEventQueue() {}

	void addButtonPressed(uint16_t buttonID)
	{
		Event e;
		e.type = Event::EventType::buttonPressed;
		e.asButtonPressed.id = buttonID;
		ScopedIrqBlocker sIrqBl;
		events_.pushBack(e);
	}

	void addButtonReleased(uint16_t buttonID)
	{
		Event m;
		m.type = Event::EventType::buttonReleased;
		m.asButtonReleased.id = buttonID;
		ScopedIrqBlocker sIrqBl;
		events_.pushBack(m);
	}

	void addEncoderTurned(uint16_t encoderID, int16_t increments, uint16_t stepsPerRev)
	{
		Event e;
		e.type = Event::EventType::encoderTurned;
		e.asEncoderTurned.id = encoderID;
		e.asEncoderTurned.increments = increments;
		e.asEncoderTurned.stepsPerRev = stepsPerRev;
		ScopedIrqBlocker sIrqBl;
		events_.pushBack(e);
	}

	void addPotMoved(uint16_t potNumber, float newPosition)
	{
		Event e;
		e.type = Event::EventType::potMoved;
		e.asPotMoved.id = potNumber;
		e.asPotMoved.newPosition = newPosition;
		ScopedIrqBlocker sIrqBl;
		events_.pushBack(e);
	}

	Event getNextEvent()
	{
		ScopedIrqBlocker sIrqBl;
		if (events_.isEmpty())
		{
			Event m;
			m.type = Event::EventType::invalid;
			return m;
		}
		else
		{
			return events_.popFront();
		}
	}

	bool isQueueEmpty()
	{
		ScopedIrqBlocker sIrqBl;
		return events_.isEmpty();
	}

private:
	FIFO<Event, 256> events_;
};
````

stephenhensley, 2020-06-30:
I'm working on a pretty minimal project that'll be a good place to try this out. 

I'll drop it into that and see how it goes. Looks like this can be a pretty clean add into the library, though.

For my initial test I'll probably scan/poll the events in the Audio Callback, and dispatch/handle the events in the main loop. Since we don't have a low priority timer callback set up yet.

TheSlowGrowth, 2020-06-30:
I have the code running already on my test board. I'm still doing the actual device drivers, but most stuff is there and ready to be tested. I'll make a PR for you to try it out.

TheSlowGrowth, 2020-06-30:
Before I migrate that into libDaisy for a PR, a quick question: I'm nut sure where to put the code. Currently stuff is organised with the filename prefixes "per_xyz", "hid_xyz" but I honestly don't know what prefix I should choose. I'd rather place things in subfolders, honestly, but with the current structure of the library, that would look wrong.
Any ideas?

stephenhensley, 2020-07-01:
Awesome! Sounds great!

As for the file naming/placement.

At first there really weren't that many files involved, and adding multiple directories does somewhat effect the simplicity of the build. However, with the HAL, CMSIS, Middlware stuff already in directories, and the increasing number of files involved in libdaisy itself, I would be open to moving to directories instead of the existing Hungarian notation (i.e. having `Peripheral/`, `Device/`, `Interface/`, `Boards/`, etc.)

But for reference, and I think there's a brief overview of this in the README of the repo, from highest to lowest level:

- `daisy_` -- high level board support files (basically an API layer)
- `hid_` -- human interface (devices), basically anything a user would interact with directly. Leds, knobs, buttons, etc.
- `dev_` -- device drivers: shift registers, led drivers. Stuff that generally uses on board peripherals to communicate with other devices.
- `per_` -- internal peripheral drivers: communication protocols/interfaces/timers

That all uses the ST HAL, which when we get around to scraping that out, would pour into the `per_` category, and a lower level category for register accesses, etc. 

And then `sys_` and `util_` are fairly self-descriptive. Though, DMA could arguably be in the `per_` group.

---

Functionally, `EventQueue` would probably be a `util`, or `hid` but could arguably fit into a new `ui` category.

stephenhensley, 2020-07-01:
Have you seen mutable instruments' [EventQueue in stmlib](https://github.com/pichenettes/stmlib/blob/master/ui/event_queue.h). Conceptually looks very similar to yours. Some minor differences, and a just a tad more abstract than what you have above. 

I haven't seen the user-side of yours yet, but I imagine its pretty similar. There are a lot of typical use-case examples in the mutable instruments Eurorack code using the stmlib EventQueue if you want to see it in use.

PR #333 (2021-06-28): Feature/UI system
This PR adds a full UI system that I've been using for various projects now.

This is the basic idea:

- hardware drivers (`PotMonitor `and `ButtonMonitor`) add events to a `UiEventQueue`
- A `UI` object has a stack of `UiPage`s to which it distributes events from the UiEventQueue.
- Multiple Canvases (represented by a `void*`) are used to draw the `UiPage`s on the actual hardware. Such a canvas is completely generic and could be a bunch of LEDs, a text display, a graphics display, etc. Inside the `UiPage` the `void*` of the canvas can be casted to whatever output device it r [... cut]
closed 2021-06-28 by stephenhensley