daisyaudio/libDaisy #206: GateIn class needs a Gate() function in addition to the Trig() function

OZHALL, 2020-06-20:
The Trig() function of the GateIn class works fine with the ADenv.h/.cpp class.  It does not  work properly with the Adsr.h/.cpp class.
I have attached a zip of hid_gatein.h/.cpp.   I have not changed any existing code.  This is simply an additional member function Gate().    I have successfully tested this.  The Gate() output accurately tracks the state of the input pin, with latency of a single sample. 
Please add this to the class in libdaisy.

[hid_gatein.zip](https://github.com/electro-smith/libDaisy/files/4807997/hid_gatein.zip)

Here is an wrapper function for calling the Gate() function for one of the elements in an array of GateIn objects:

bool DaisyOctopot::GetGateInGate(int iGateNumber)
{
    return gate_input[iGateNumber].Gate();
}

here is an example of retrieving the gate and passing it to the Adsr::Process() function:
    static Adsr       env;
...
        // Use envelope to control the amplitude of the VCA (right out).
        gate       = hw.GetGateInGate(DaisyOctopot::GATE_IN_2);
        env_out    = env.Process(gate);

stephenhensley, 2020-06-22:
Added just before you opened the issue, though its called `State(`) instead of `Gate()`.
#### State
Checks current state of gate input (no state required)

read function is inverted because of suggested BJT input circuit

```C++
    inline bool State() { return !dsy_gpio_read(&pin_); }
```

closed 2020-06-22 by stephenhensley

OZHALL, 2020-06-23:
Thx Stephen

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus