daisyaudio/libDaisy #216: Building on PlatformIO

wraybowling, 2020-07-02:
I've started the Daisy Patch verb example using PlatformIO and kept the makefile in place for good measure. I get various build failures from either method. Here, you can see I've set up both workflows with Github actions though they may be configured incorrectly. https://github.com/MicroWrave/daisy-quad-fx/actions I'm trying to set this repo up as a proof of concept for building daisy code with PlatformIO and/or actions so that failing builds do not merge into releases.

When building via PlatformIO, these are the errors
```
> Executing task: platformio run <

Processing electrosmith_daisy (platform: ststm32; board: electrosmith_daisy; framework: stm32cube)
----------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/electrosmith_daisy.html
PLATFORM: ST STM32 7.0.0 > Electrosmith Daisy
HARDWARE: STM32H750IBK6 400MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-stm32cube 2.0.181130 
 - tool-ldscripts-ststm32 0.1.0 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 24 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <libdaisy>
|-- <DaisySP>
Building in release mode
Compiling .pio/build/electrosmith_daisy/libd51/libdaisy/fatfs.o
Compiling .pio/build/electrosmith_daisy/libd51/libdaisy/hid_audio.o
In file included from lib/libdaisy/src/fatfs.c:1:0:
lib/libdaisy/src/fatfs.h:9:10: fatal error: ff.h: No such file or directory

************************************************************
* Looking for ff.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ff.h"
* Web  > https://platformio.org/lib/search?query=header:ff.h
*
************************************************************

 #include "ff.h"
          ^~~~~~
compilation terminated.
Compiling .pio/build/electrosmith_daisy/libd51/libdaisy/hid_usb.o
Compiling .pio/build/electrosmith_daisy/libd51/libdaisy/hid_wavplayer.o
Compiling .pio/build/electrosmith_daisy/libd51/libdaisy/per_gpio.o
lib/libdaisy/src/hid_usb.cpp:2:10: fatal error: usbd_core.h: No such file or directory

*******************************************************************
* Looking for usbd_core.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:usbd_core.h"
* Web  > https://platformio.org/lib/search?query=header:usbd_core.h
*
*******************************************************************

 #include "usbd_core.h"
          ^~~~~~~~~~~~~
compilation terminated.
In file included from lib/libdaisy/src/hid_wavplayer.cpp:3:0:
lib/libdaisy/src/fatfs.h:9:10: fatal error: ff.h: No such file or directory

************************************************************
* Looking for ff.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ff.h"
* Web  > https://platformio.org/lib/search?query=header:ff.h
*
************************************************************

 #include "ff.h"
          ^~~~~~
[... 98 lines cut ...]
                 from lib/libdaisy/src/per_uart.cpp:1:
lib/libdaisy/src/per_uart.cpp: In function 'void HAL_UART_MspInit(UART_HandleTypeDef*)':
/Users/wraybowling/.platformio/packages/framework-stm32cube/h7/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_rcc.h:1394:48: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
                                         UNUSED(tmpreg); \
/Users/wraybowling/.platformio/packages/framework-stm32cube/h7/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_def.h:88:27: note: in definition of macro 'UNUSED'
 #define UNUSED(x) ((void)(x))
                           ^
lib/libdaisy/src/per_uart.cpp:174:9: note: in expansion of macro '__HAL_RCC_USART1_CLK_ENABLE'
         __HAL_RCC_USART1_CLK_ENABLE();
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/wraybowling/.platformio/packages/framework-stm32cube/h7/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_rcc.h:935:48: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
                                         UNUSED(tmpreg); \
/Users/wraybowling/.platformio/packages/framework-stm32cube/h7/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_def.h:88:27: note: in definition of macro 'UNUSED'
 #define UNUSED(x) ((void)(x))
                           ^
lib/libdaisy/src/per_uart.cpp:176:9: note: in expansion of macro '__HAL_RCC_GPIOB_CLK_ENABLE'
         __HAL_RCC_GPIOB_CLK_ENABLE();
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/electrosmith_daisy/libd51/libdaisy/per_uart.o] Error 1
======================================= [FAILED] Took 1.83 seconds =======================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

stephenhensley, 2020-07-06:
It looks like the libdaisy Driver/Middleware dependencies are not being included (the STM32 HAL, and USB/FatFS middlewares).

I'm pretty unfamiliar with PlatformIO, but I can check it out a bit, and report back later with any ideas.

wraybowling, 2020-07-06:
fwiw, I'm on MacOS. I followed the guide to build with Make, and got similar errors. I did try using git submodules and updating recursively. In the repo linked, I've moved on to trying platformIO's own package manager to see if that would help, but it didn't make a difference. Jump back to an earlier commit to try that. Once we get this working, feel free to copy it as a third option to building daisy projects. I was just trying to help out!

stephenhensley, 2020-07-06:
@wraybowling If you were seeing build errors with Make it would be helpful if you could share them (even if they're similar to above). That would at least give me somewhere to begin with trying to replicate the issue on mac before trying to do it with platformIO.

wraybowling, 2020-07-07:
Perhaps not so similar. This is from the verb directory of the DaisyExamples repo at f222fc5

```
gmake
arm-none-eabi-gcc -c -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32H750xx -DUSE_HAL_DRIVER -DSTM32H750xx -I../../libdaisy/core/ -I../../libdaisy -I../../libdaisy/src/ -I../../libdaisy/Drivers/CMSIS/Include/ -I../../libdaisy/Drivers/CMSIS/Device/ST/STM32H7xx/Include -I../../libdaisy/Drivers/STM32H7xx_HAL_Driver/Inc/ -I../../DaisySP  -O2 -Wall -fdata-sections -ffunction-sections -MMD -MP -MF"build/system_stm32h7xx.d" -Wa,-a,-ad,-alms=build/system_stm32h7xx.lst ../../libdaisy/core/system_stm32h7xx.c -o build/system_stm32h7xx.o
arm-none-eabi-g++  -c -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32H750xx -DUSE_HAL_DRIVER -DSTM32H750xx -I../../libdaisy/core/ -I../../libdaisy -I../../libdaisy/src/ -I../../libdaisy/Drivers/CMSIS/Include/ -I../../libdaisy/Drivers/CMSIS/Device/ST/STM32H7xx/Include -I../../libdaisy/Drivers/STM32H7xx_HAL_Driver/Inc/ -I../../DaisySP  -O2 -Wall -fdata-sections -ffunction-sections -MMD -MP -MF"build/ex_verb.d" -fno-exceptions -fasm -finline -finline-functions-called-once -fshort-enums -fno-move-loop-invariants -fno-unwind-tables  -Wa,-a,-ad,-alms=build/ex_verb.lst ex_verb.cpp -o build/ex_verb.o
arm-none-eabi-g++  build/system_stm32h7xx.o build/ex_verb.o build/startup_stm32h750xx.o -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=hard --specs=nano.specs --specs=nosys.specs -T../../libdaisy/core/STM32H750IB_flash.lds -L../../libdaisy/build  -L ../../DaisySP/build -ldaisy -lc -lm -lnosys -ldaisysp -Wl,-Map=build/ex_verb.map,--cref -Wl,--gc-sections -o build/ex_verb.elf
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot find -ldaisy
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot find -ldaisysp
collect2: error: ld returned 1 exit status
gmake: *** [../../libdaisy/core/Makefile:189: build/ex_verb.elf] Error 1
```

stephenhensley, 2020-07-13:
@wraybowling  Sorry for the delay.

I installed PlatformIO today, and should be able to get to testing building the Daisy stuff this week, but the last error you posted looked like you just didn't build the libraries first. That can be done by running `make` in the `libdaisy/` and `DaisySP/` directories, or by running the `rebuild_libs.sh` script.

However if you were having build issues with the libraries, that would explain why it can't find the libs to link to. I should have more information in the next few days.