daisyaudio/libDaisy #233: make program can't find interface/stlink.cfg

wraybowling, 2020-07-18:
I successfully built an example. now i'm trying to flash daisy using the instructions for an ST-Link 2 and openocd.

```
make program
openocd -s /usr/local/share/openocd/scripts -f interface/stlink.cfg -f target/stm32h7x.cfg \
	-c "program ./build/ex_verb.elf verify reset exit"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:60: Error: Can't find interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
make: *** [../../libdaisy/core/Makefile:230: program] Error 1

```

stephenhensley, 2020-07-27:
Depending on where you got your openocd it may not have the standard stlink.cfg yet (I believe that file is a combo for the v1, v2, and v3 that recently came out).

You could try changing `PGM_DEVICE` to `interface/stlink-v2.cfg`

wraybowling, 2020-08-03:
I can dig deeper later as to which exact version of openocd i got my hands on, but I can say it was whatever Ubuntu Buster distributes in their apt listing. Very likely 0.10.0 as I installed it while following the wiki steps.

stephenhensley, 2020-08-13:
Sorry for the late response, there are several versions of openocd available that are 0.10.0, unfortunately many of them are older than what can be used for the stm32h750.

I have recently tried out the version distributed on xpack that was discussed on the forum, and had success using it.

Here are the install instructions for that (it should work for all OSes), they both manual install instructions, and instructions for installing via npm.

https://xpack.github.io/openocd/install/

We'll add references to these instructions to the Wiki shortly!

stephenhensley, 2020-08-13:
Let me know if this works out for you so we can close the issue 😄

wraybowling, 2020-08-14:
Made time to get to my workbench today. First i'll check my openOCD version

```
$ openocd --version
Open On-Chip Debugger 0.10.0
```

Then pulling latest from git...
```
$ git pull && git submodule update -r
(blah blah blah)
```

then rebuilding...
```$ ./rebuild_all.sh
(blah blah blah)
done
finished
```

Navigating to the FilterBank example...
`cd patch/FilterBank`

Refreshing wiki...
https://github.com/electro-smith/DaisyWiki/wiki/5.-How-To-Flash
which links to [a mouser page that matches the ST-LINK/V2 programmer I'm using](https://www.mouser.com/ProductDetail/STMicroelectronics/ST-LINK-V2?qs=sGAEpiMZZMu3sxpa5v1qrqPIYHPyeOmOUBuCAg6HF1c%3D).

Running like before...
```$ make program
openocd -s /usr/local/share/openocd/scripts -f interface/stlink.cfg -f target/stm32h7x.cfg \
	-c "program ./build/FilterBank.elf verify reset exit"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:60: Error: Can't find interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
make: *** [../../libdaisy/core/Makefile:241: program] Error 1
```

looking for PGM_DEVICE in the examples root...
```$ ack PGM_DEVICE
libdaisy/core/Makefile
18:PGM_DEVICE ?= interface/stlink.cfg
19:OCDFLAGS = -f $(PGM_DEVICE) -f target/$(CHIPSET).cfg
```

Changed line 18... and saved
```PGM_DEVICE ?= interface/stlink-v2.cfg```

Retrying...
```
$ ./rebuild_all.sh
( etc... )
$ cd patch/FilterBank
$ make program
openocd -s /usr/local/share/openocd/scripts -f interface/stlink-v2.cfg -f target/stm32h7x.cfg \
	-c "program ./build/FilterBank.elf verify reset exit"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:60: Error: Can't find target/stm32h7x.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
make: *** [../../libdaisy/core/Makefile:241: program] Error 1
```

fwiw, yes I pressed boot->reset on the daisy seed. The daisy and ST programmer both have a steady red light.

wraybowling, 2020-08-14:
Following up, i'll try the openocd package you're talking about

installing npm
`sudo apt-get install npm`

uninstalling openocd
`sudo apt-get --purge remove openocd`

installing xpm
`sudo npm install --global xpm@latest`

installing openocd@latest
`xpm install --global @xpack-dev-tools/openocd@latest`

try programming
```
$ make program
openocd -s /usr/local/share/openocd/scripts -f interface/stlink-v2.cfg -f target/stm32h7x.cfg \
	-c "program ./build/FilterBank.elf verify reset exit"
/bin/sh: 1: openocd: not found
make: *** [../../libdaisy/core/Makefile:241: program] Error 127
```

check that it's there
```
$ which openocd
```
(nothing)

Found it in...
```
$ cd /${HOME}/opt/xPacks/@xpack-dev-tools/openocd && ls
0.10.0-14.3
```
but it's not installed per se...

stephenhensley, 2020-08-14:
if its missing the `target/stm32h7x.cfg` then its definitely an old version of openocd (every version, from everyone that maintains a separate build for the past year to two years has been 0.10.0, which is longer than the stm32h7x chips have been around).

I was about to suggest installing the xpack one, but you just did that. I haven't personally used xpm/npm to install it yet, I've only done a manual installation (download the files, and add the directories `bin/` folder to my `PATH` variable.

Looks like you should be in business if you add `${HOME}/opt/xPacks/@xpack-dev-tools/openocd/bin` to your `PATH` in bashrc or elsewhere. Or xpack might have taken care of something under the hood that requires a restart or at least reopening the terminal.

Hope that helps! Sorry you've had such a tricky time getting it up and running.

wraybowling, 2020-08-14:
check that it's there
```
$ which openocd
```
(nothing)

Found it in...
```
$ cd /${HOME}/opt/xPacks/@xpack-dev-tools/openocd && ls
0.10.0-14.3
```
but it's not installed per se... 

From the install notes...
>  Important: Automatic path discovery for the packages from the new @xpack-dev-tools scope was added to GNU MCU Eclipse plug-ins with v4.6.1-201909231407; update older versions or configure the path manually.

Well that's bullcrap. *xpm, you had one job.* I do not recommend xpm. I'll try a manual method.

wraybowling, 2020-08-14:
As for the inconvenience.. *loud sigh! ugh!* It's okay. No pain, no gain. Somehow, through this, we'll find sticking points and the wiki will be better for it. :P

okay! downloading distribution directly here
https://github.com/xpack-dev-tools/openocd-xpack/releases/

unpacked it to a `Developer` folder with a generic name in case there's a version bump later
`~/Developer/xpack-openocd`

changed my ~/.bashrc to include that path's `bin` directory... so now it looks like this (ymmv)
```
# Wray's stuff
GCC_PATH=~/Developer/gcc-arm-none-eabi-9-2020-q2-update/bin
OpenOCD_PATH=~/Developer/xpack-openocd/bin
export PATH=$GCC_PATH:$OpenOCD_PATH:$PATH
```

Quit and restart the terminal because i can never remember how to do that "source" thing that reloads settings :P

We in business, boss
```
$ which openocd
/home/wray/Developer/xpack-openocd/bin/openocd
```

try programming........!
```
$ make program
openocd -s /usr/local/share/openocd/scripts -f interface/stlink-v2.cfg -f target/stm32h7x.cfg \
	-c "program ./build/FilterBank.elf verify reset exit"
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1800 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.181028
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000c3c msp: 0x20020000
** Programming Started **
Info : Device: STM32H74x/75x
Info : flash size probed value 128
Info : STM32H7 flash has a single bank
Info : Bank (0) size is 128 kb, base address is 0x8000000
Info : Padding image section 1 at 0x0800dd5c with 4 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x0800dd60 .. 0x0801ffff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```

wraybowling, 2020-08-14:
I pronounce this problem...

[image]

...SOLVED! 🎸

closed 2020-08-14 by wraybowling

stephenhensley, 2020-08-15:
@wraybowling awesome!

Glad you got it sorted!

I'll adjust the wiki sometime this weekend to describe these steps in detail. I'll probably just set up a dedicated 'debugging/programming with openocd' page since it will be the same/similar process for each OS.