daisyaudio/libDaisy #234: Build fails due to warning in UNUSED macro

tvannoy, 2020-07-26:
When building libDaisy, I get the following output from `make`:
```c
In file included from Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_rcc.h:29:0,
                 from src/stm32h7xx_hal_conf.h:268,
                 from Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal.h:30,
                 from src/per_adc.cpp:1:
src/per_adc.cpp: In function 'void HAL_ADC_MspInit(ADC_HandleTypeDef*)':
Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_rcc.h:803:48: error: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}' [-Werror]
                                         UNUSED(tmpreg); \

Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_def.h:70:27: note: in definition of macro 'UNUSE '
 #define UNUSED(x) ((void)(x))
                           ^
src/per_adc.cpp:469:9: note: in expansion of macro '__HAL_RCC_ADC12_CLK_ENABLE'
         __HAL_RCC_ADC12_CLK_ENABLE();
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
Makefile:258: recipe for target 'build/per_adc.o' failed
make: *** [build/per_adc.o] Error 1
```

The build works if I remove `-Werror` from the makefile. 

This issue appears to be related to #216. The output from `make` reports the same error in that issue as well. 

## Environment
I'm using Windows Subsystem for Linux (WSL) on Windows 10.
I've installed arm-none-eabi from the Ubuntu repository on WSL; arm-none-eabi is version `arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620`

--- 

I don't know enough about the HAL code to know if the problem is actually the UNUSED macro or if the problem is with my toolchain/environment. If "conversion to void" warning does not show up for other people, I'm guessing #216 and my toolchain have differences from other people's.

stephenhensley, 2020-07-27:
I vaguely recall these warnings from a while ago, but they don't appear in my build on Windows anymore. I'll test on a linux build as well, and/or remove `-Werror` from the HAL portion of the build.

tvannoy, 2020-07-27:
I just testing on a Ubuntu 20.04 with `gcc-arm-none-eabi` installed via apt. I don't get those warnings in that case. My arm-none-eabi version in this case is `arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]`. 

I suspect the problem is just that the version of `arm-non-eabi-gcc` that is included in Ubuntu 18.04 in WSL is too old. It's two years old compared to what's included in Ubuntu 20.04.

stephenhensley, 2020-07-27:
Could be. I've seen a few different varieties of errors/warnings come and go over the years.

On my current machines I have tools ranging from `8-2019-q3` to the most recent `9-2020-q2`. I don't think I've tried to build libdaisy with anything older than that at this point.

tvannoy, 2020-07-28:
I think we can close this issue. Maybe it would be good to include some info in the wiki about supported compiler versions.

erwincoumans, 2020-08-01:
The issue is still there, also got a warning, resulting in an error on Mac:
~~~~~~~~
![image](https://user-images.githubusercontent.com/725468/89110899-75d3e480-d404-11ea-9a94-6a3a903b866e.png)

~~~~~~~~

tvannoy, 2020-08-02:
@erwincoumans  Okay, good to know it wasn't just me. Out of curiosity, what version of the arm-none-eabi toolchain are you running? 

There is a [stackoverflow post](https://stackoverflow.com/questions/49061212/stm32-hal-library-warning-with-c14-above) with more information about this warning in the STM32 HAL. There's quite a bit of good information in the post. According to one user, this warning has been fixed in newer versions of GCC, which agrees with the behavior I saw.

In the post, there are two suggestions on how to avoid this warning, the latter of which has been used by STMicro on another processor: 
```c
#include "stm32f3xx_hal_def.h"
#undef UNUSED
#define UNUSED(x) ((void)((uint32_t)(x)))
```
or 
```c
#include "stm32f3xx_hal_def.h"
#undef UNUSED
#define UNUSED(x) (void)x
```

catfact, 2020-08-08:
just as another data point:  i am also encountering this warning/error building on ubuntu 18.04.4 (latest LTS as of this writing and probably most common linux distro.)

compiler version:
```
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
```

package info for `gcc-arm-none-eabi` which provides it:
[details]
```
Package: gcc-arm-none-eabi
Architecture: amd64
Version: 15:6.3.1+svn253039-1build1
Priority: extra
Section: universe/devel
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Agustin Henze <tin@debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 300080
Depends: libc6 (>= 2.15), libgcc1 (>= 1:3.0), libgmp10, libisl15 (>= 0.15), libmpc3, libmpfr6 (>= 3.1.3), libstdc++6 (>= 5), zlib1g (>= 1:1.1.4), binutils-arm-none-eabi
Recommends: libnewlib-arm-none-eabi
Filename: pool/universe/g/gcc-arm-none-eabi/gcc-arm-none-eabi_6.3.1+svn253039-1build1_amd64.deb
Size: 24317260
MD5sum: 69275ead19d4e87539a3a78e965062f9
SHA1: 6f7158a58c27bb368e45c65dab4a175192ecbc0a
SHA256: ff0af32a53aa8023eb7a5df6e7b2dfa8b24ea6e338a669dfbdeb92ffade3c15f
Homepage: http://gcc.gnu.org/
Description-en: GCC cross compiler for ARM Cortex-A/R/M processors
 Bare metal compiler for embedded ARM chips using Cortex-M, Cortex-R and
 Cortex-A processors.
 This package is based on the GNU ARM toolchain prov
```
[/details]

two workarounds:
- in `libdaisy/Drivers/STM32H7xx_HAL_Driver/Inc/`, changing [L70](https://github.com/electro-smith/libDaisy/blob/master/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_def.h#L70) to:
`#define UNUSED(x) ((void)((uint32_t)(x)))`
- of course just disabling `-Werror` in `libdaisy/Makefile`

i may try with a more bleeding-edge version of arm-gcc toolchain next time i am on a distro which better supports it.

tvannoy, 2020-08-08:
It certainly seems like a compiler version issue. The Daisy Wiki tends to suggest downloading the ARM toolchain directly from ARM's website, so people who follow those instructions shouldn't have any problems. However, there are enough of us using the toolchain provided by our package managers, so it seems worthwhile to implement one of the suggested workarounds. 

Unless somebody beats me to it, I will try to find time to make a PR to fix this issue this coming week. Any thoughts on which workaround we should implement?

erwincoumans, 2020-08-08:
Any solution that avoids the error is fine with me. I just removed the -Werror for now.

stephenhensley, 2020-08-13:
`-Werror` has been disabled for the time being. 

I saw this as well when running with the version of arm-none-eabi-gcc (5.4.1 20160919) that was in the default debian package manager.

That said, the Wiki does specify using a more modern version of the compiler, which also prevents these warnings.

Since it can be built without error-ing out now, I'll close this.

closed 2020-08-13 by stephenhensley