daisyaudio/libDaisy #238: Excess time spent in AudioCallback causes dropped MIDI bytes

hammondeggs, 2020-07-27:
Currently, if you spend too much time in your AudioCallback routine, the MIDI UART handler will occasionally drop bytes.

You can reproduce this with the existing Daisy Pod Midi example by having the CPU perform 'busy work' (for example a large FIR or other dummy calculations if you like) during the callback.

For now, I am using a temporary workaround where I simply play a buffer in the audio callback and build this buffer outside any interrupts / DMA transfers as needed. This of course isn't ideal, but it does work, and barely uses any CPU time. 

(For what it's worth I had a look at libdaisy and tried inverting the DMA priority of the UART DMA and the audio DMA but no luck).

stephenhensley, 2020-07-27:
Good to know! 

I want to look into the UART peripheral and how it interfaces with the DMA a bit more, because it would be ideal to set it up to be able to fill a fairly large FIFO, and still be able to be read with as few as 1 or 2 bytes. 

Right now the hardware is trying to throw an interrupt for every byte received so that it can fill a software FIFO without stalling when there isn't a full buffer.

This will be happening along with some other MIDI-related stuff over the next week or so.

stephenhensley, 2020-09-04:
I'm going to address this in #250 as well as the running status fix. 

I have a mostly fixed system in place that augments the DMA use to read up to 256 bytes, and uses the IDLE line interrupt to trigger the buffer copy (this interrupt is fired when there is no activity on the bus for the 1 byte-period). 

I'm currently testing, and cleaning up, but it's working very reliably.

stephenhensley, 2020-09-10:
This was fixed in #250

closed 2020-09-10 by stephenhensley

andrewikenberry, 2020-10-28:
[image] [Flexible SPI interface (all configurations for Rx and Tx)](https://trello.com/c/tSEWp74r/16-flexible-spi-interface-all-configurations-for-rx-and-tx)