daisyaudio/libDaisy #239: MIDI does not appear to support Running Status / NoteOff via NoteOn

hammondeggs, 2020-07-27:
I know the MIDI implementation is currently incomplete however I will note a few things anyway in case they are helpful.

Currently, the MIDI implementation appears to require the full 3 byte Status, msg, msg format. E.g. NoteOn, NoteNum, Velocity etc.

To save time, MIDI supports something called "Running Status" where you can send a single NoteOn message followed by pairs of bytes for Note Num / Velocity - allowing you in this example to send multiple note ons (*and possibly note offs as well explained later).

Note as well, that a NoteOn message with a velocity of "'0x00'" is actually processed as a NoteOff (with a velocity of zero). This also means that during a running status message you can have both note ons and note offs.

For example:

'0x90 3C 4A 3D 00 3E 4A'

3 messages are present here in this one group: 
'0x90' - Note On, Channel 1
'0x3C' - Note Num for note on
'0x4A' - Velocity for this note on
'0x3D' - Next Note num (still using '0x90' status)
'0x00' - Velocity is 00 - this should be treated exactly the same as `0x80 0x3D 0x00`
'0x3E' - Next Note num (still using '0x90' status)
'0x4A' - velocity for this note num.

Etc.

stephenhensley, 2020-07-27:
Thanks for the feedback! This is helpful, we'll be digging into expanding the MIDI handling quite a bit over the next week or so.

chrisfromwork, 2020-09-02:
Is someone actively working on midi support or would it be worth looking into these issues? I recently picked up a patch and haven't been able to observe any midi messages when sending midi through the trs input from a keystep pro.

chrisfromwork, 2020-09-02:
talking with folks on the forum it seems like my issues may be related to having a type B trs to midi adapter. ill test out a type A and see if i still hit the same issues

chrisfromwork, 2020-09-04:
Type A midi seems to register events, but i see note on and note off skipping. Is someone actively working on midi?

stephenhensley, 2020-09-04:
@chrisfromwork I had to jump on a few other things for a bit, but I'll be tackling this.

Also, the boards are made to work with Type A TRS MIDI. Type B MIDI isn't supported by default (However, there are some 0R resistors that can be moved on the bottom of the Daisy Patch to support type B instead).

chrisfromwork, 2020-09-05:
@stephenhensley thanks for the response!

PR #250 (2020-09-08): added running status support to MIDI
This should fix #239 

Most of my hardware doesn't seem to use running status, but I was able to find an older oxygen8 that I couldn't reliably use with the MIDI examples before this update, and was actually using the running status feature. 

Testing from a few sources this seems to fix a lot of the 'dropped note' issues. However, I am still finding that if too much time is spent in the audio callback this will cause UART to miss new data coming in. So I'll reopen the issue discussing that problem as we'll probably we want to flesh out thata implementation [... cut]
closed 2020-09-08 by stephenhensley