daisyaudio/libDaisy #241: SD Card does not initialize 100% of the time

hammondeggs, 2020-08-05:
Currently, with an "older" Kingston 8GB SDHC card (model SDC4/8GB 07), I am noticing that the card is not recognized every time on powerup.

In `per_sdmmc.cpp`, at `SdmmcHandler::Init()` , we see a couple lines:

```
    //hsd1.Init.ClockDiv = 168; // 476.2kHz works    
    hsd1.Init.ClockDiv = 6; // 12MHz works 
```

I un-commented the 476KHz clock divider and commented out the 12Mhz clock div, rebuilt libdaisy and sure enough, the card appears to initialize every time.

Now, I know that a 476KHz SD card read speed isn't ideal (although for what I'm currently doing it's perfectly fine), so there likely is a happy medium here that would probably have to be deduced via some testing with different cards etc.

(FWIW I had a look at the traces on a scope, at 12mhz they're square enough but there appears to be some ~12MHz noise on the high level of the signal, not sure if this is related to this issue).

stephenhensley, 2020-08-24:
I think I had noticed this occasionally as well, but most of the time it was by trying to mount the card while Audio, and other high-priority interrupts had been enabled.

I remember the same bug as @upbeat27, but I believe it was fixed in one of the (semi)-recent HAL updates. Looking at the HAL, it seems to boot up the SD card at a fixed rate 476kHz, and then, updates the prescaler to the user defined value after initialization (might be worth actually verifying this, though). 

I spent a good chunk of time last week with the SDMMC peripheral on the H7. I'll be updating the driver here a bit later this week. I was able to get it running with STs DMA template , and working up to 50MHz (I did not test writing w/ DMA or at the higher speeds yet).

stephenhensley, 2021-01-18:
In addition to the above, it was pointed out that changing the `GPIO_InitStruct.Speed` value to `LOW` helps with the stability. We'll change that while reworking the rest of the driver.

stephenhensley, 2021-03-10:
this should be resolved with the merging of #311 

There is a note in there about the GPIO speed, and I can open another PR that adds that to the config.

closed 2021-03-10 by stephenhensley