daisyaudio/libDaisy #247: Calibrating CVs and storing on QSPI

grrrwaaa, 2020-08-28:
The CV ins & outs are not 100% linear, so for certain applications (e.g. v/oct IO) some kind of calibration is needed. 
(A quick test on my Patch shows the CV input to drift by about half a semitone over 3 octaves).

The results of calibration could be stored on persistent flash, and used (optionally/automatically?) if found there on boot, to ensure voltage precision. Having a standardized layout for calibration data would make this portable & reusable between different binaries. 

Not clear whether a simple scale/offset, a multi-breakpoint, a polynomial type fit or something else would be most appropriate for the calibration data.

grrrwaaa, 2020-08-28:
Actually, going up octaves on the keyboard I see non-uniform steps of 1.009, 2.027, 3.045, 4.065, so it doesn't look like a simple scale/offset alone would be sufficient.

jbeda, 2020-11-01:
I created a calibration mode for that kicks in when you hold down the encoder on boot. Also saves some settings to QSPI flash.  I'm just tracking min/max values but the idea would be that you could do further calibration for curves.

Feel free to steal/borrow the code.  If we can get something good together we can extract into a library and perhaps add it ot the examples repo.

https://github.com/jbeda/daisy-experiments/commit/4f63548131dfaa012c2f2842d54e128a36425820

jbeda, 2020-11-01:
Ok -- bit the bullet and created a piece wise calibration mode.  UI might need to be tweaked.  If you boot up with the encoder held down it'll go into a mode where you can put in reference voltages for each CV at 1V steps. Press the encoder again to exit calibration mode. These are then saved into QSPI and there is a helper to read the calibrated value in volts.

I found that I get different readings if audio isn't started yet so this starts audio and runs an oscillator out on each output.

https://github.com/jbeda/daisy-experiments/commit/c465050e84ea2da98783105bcacefab3e1ee05e4

stephenhensley, 2021-09-30:
oh hey. This is now in v2.0.0 😄

closed 2021-09-30 by stephenhensley