daisyaudio/libDaisy #268: Audio I/O channels swapped with certain build settings

stephenhensley, 2020-11-17:
This is particularly evident when switching between the default Make build and the VS/VisualGDB build.

It seem like a timing issue related to the SAI StartDMA function, as adding a delay in the middle of that function can swap the channels back. 

It also seems like this changes based on debug/release settings like optimization.

I'll add more details as I investigate further

TheSlowGrowth, 2021-04-29:
This sounds like a reordering problem to me. During the optimization, the compiler may be reordering access to the various registers and swap thing around. Have you tried adding the `__sync_synchronize();` function as a barrier?

stephenhensley, 2021-04-29:
It's been a while since I was tinkering on this, but I'm fairly certain I tried all of `__sync_synchronize()`, `__DSB()`, etc.

What was also interesting was the length of the delay that I had inserted in the `HAL_SAI_StartDMA` seemed to have an effect on the channel offset. It really seems like the data lines are getting offset from the FS lines by a fixed number of bits that seems related to the timing between enabling the SAI and starting the transaction.

There is this note:

>Each of the audio blocks in the SAI are enabled by SAIEN bit in the SAI_xCR1 register. As soon as this bit is active, the transmitter or the receiver is sensitive to the activity on the clock line, data line and synchronization line in slave mode. In master TX mode, enabling the audio block immediately generates the bit clock for the external slaves even if there is no data in the FIFO, However FS signal generation is conditioned by the presence of data in the FIFO. After the FIFO receives the first data to transmit, this data is output to external slaves. If there is no data to transmit in the FIFO, 0 values are then sent in the audio frame with an underrun flag generation.

From the reference manual (rev7 page 2259),

That seems like it could be related, 

---

Thanks for bumping this, though. I've totally been neglecting it, and it should really get addressed sooner than later.

stephenhensley, 2021-10-15:
Fixed in #420