daisyaudio/libDaisy #277: Can't get ADC working with audio callback

ThomArmax, 2020-12-07:
Hi,

I'm very new to Daisy, I have just received my Seed a few days ago.

Basically, I'm trying to modify the oscillator frequency using a potentiometer. But the read of ADC always returns the same value whatever the pot position...

I have tested the Knob example from Web programmer, and it works.

If i comment the line `seed.StartAudio(AudioCallback);` it works. (PS I have tested with empty `AudioCallback`, same result, doesn't work)

I'm using
- libDaisy@`c5c1026` like in DaisyExample repo
- DaisySP@`5a6b618` like in DaisyExample repo
- STM32 Cube IDE

The Cube project was created doing:
- File->New Project -> STM32 project from Existing STM32CubMx Configuration file
- Using `Daisy_Seed_Rev2.ioc` from [Daisy_Seed_Rev4 examples repo](https://github.com/electro-smith/DaisyExamples/tree/master/cube/Daisy_Seed_Rev4)
- Removed all STM32 imported files (since present in libDAisy)
- clone libDaisy and DaisySP

Do you have experienced such issue ?

EDIT : I also have the issue if read the ADC in the AudioCallback

Here is my code :

```cpp
int main(void)
{
    // initialize seed hardware and oscillator daisysp module
    float sample_rate;
    seed.Configure();
    seed.Init();
    sample_rate = seed.AudioSampleRate();

    // Oscillators init
    for (int i = 0; i < OSC_COUNT; i++)
    {
        freqs[i] = BASE_FREQ + i*FREQ_INC_FACTOR;
        oscs[i].Init(sample_rate);
        oscs[i].SetWaveform(Oscillator::WAVE_RAMP);
        oscs[i].SetFreq(freqs[i]);
        oscs[i].SetAmp(0.5);
    }

    // ADC init
    AdcChannelConfig adcConfig;
    adcConfig.InitSingle(seed.GetPin(21));
    seed.adc.Init(&adcConfig, 1);
    seed.adc.Start();

    // start callback
    seed.StartAudio(AudioCallback);

    volatile bool ledState = false;

    while(1)
    {
        freqModifier = seed.adc.GetFloat(0);

        seed.SetLed(ledState);
        ledState = !ledState;
        dsy_system_delay(100);
    }
}
```

stephenhensley, 2020-12-07:
Just a few question on the hardware side of things first (though, if the knob example is working then this is likely not the issue):

Do you have your AGND pin connected to the DGND pin on your circuit?

---

Hmm, your code does look fine to me.

Is it the same value every time, or is it an accurate value of the input at first, and then it no longer updates?

Also, are you using a debugger from the CubeIDE or just programming via USB?

I haven't personally set up a libdaisy project using CubeIDE, but can probably advise some places for breakpoints to get a bit more info if you have debugging working.

ThomArmax, 2020-12-07:
I use the same schematics as in the Knob exemple.
I use Cube IDE with JLink Edu mini SWD debug probe to program and débug application.
When I break just After ADC read, I always get the same value (65535 if normal get, not float). If i comment the startAudio, the values are fine (evoluate according to pot position)

stephenhensley, 2020-12-07:
Okay, a  couple of thoughts, based on the constant 65535 value it seems like the  linker issue, but I'm not sure why it would work correctly if audio is not started.

**Check ADC Interrupt/IRQ Handler**

Are you able to break in the `HAL_ADC_ConvCpltCallback()` function within `libdaisy/per/adc.cpp` while the audio callback is running?

**Linker Script Usage**

I'm not familiar with STM32Cube, but if it doesn't use a Makefile (and is using internal build configuration). You will have to set the Linker script to the one located in the core/ directory: `path/to/libdaisy/core/STM32H750IB_flash.lds`

Otherwise, the section of memory designated to the DMA Buffers (for ADCs and Audio) would not be configured to be cache-able. This would likely mess up any DMA Receptions (your Audio output might still work, but input data from both Audio and the ADCs would be wrong, or unchanging).

You can confirm that this is cache related by commenting out line 83 of `libdaisy/src/sys/system.c` With the DCache not enabled you'll have worse memory  performance but DMA would function. Linking to the proper linker script should resolve this without having to modify any libdaisy files.

ThomArmax, 2020-12-07:
@stephenhensley also, wich dev env should I use ?
I was thinking I would be able to work with official, free, cross platforms ST tools.
- I have not yet been able to build the libs with Cube. 
- Building manually the libs with provided scripts and developping with Cube looks like to have strange behavior
- Trying VS code with Platform IO, but could not build yet le libs too.
- Arduino is not an option, you know debug, IDE, ...
- Visual Studio with a non free compiler, not an option too (not free, not cross platforms)

> Are you able to break in the HAL_ADC_ConvCpltCallback() function within libdaisy/per/adc.cpp while the audio callback is running?

Currently not able to break into libDaisy. Sources not linked to lib. Maybe I should build the libs in debug and find how to add extra sources folder for debug

> Linker Script Usage

I use the linker script provided with libDaisy.

ThomArmax, 2020-12-07:
Maybe I have something odd in my project configuration. I'll check that with the info you gave me 👌. Maybe this could be a future improvment to have a project template for PlatformIo and/or Cube IDE 🙂

stephenhensley, 2020-12-07:
We'll be adding a lot of support for the free, cross platform VS Code environment (using Cortex Debug extension) soon. 

What kind of 'strange behavior' are you seeing when building libs manually with the provided scripts/Makefiles?

Also, since your project appears very simple for the time being. Is it possible to build your example via Make instead of using the STM32CubeIDE just to sanity check whether it's an issue within libdaisy, your project, or the CubeIDE configuration.

Once we have decent tutorials/examples for using VS Code, I plan on adding VisualStudio/VGDB and STM32CubeIDE project files to the libraries, and having at least one example. But I don't have an exact timeline on that. I also still need to spend some time looking into PlatformIO as I have not done anything with it yet.

ThomArmax, 2020-12-07:
> We'll be adding a lot of support for the free, cross platform VS Code environment (using Cortex Debug extension) soon.

Great news !

> What kind of 'strange behavior' are you seeing when building libs manually with the provided scripts/Makefiles?

This was more like about the issue I have, but probably the way I link the libs or some build flags ?

> Is it possible to build your example via Make instead of using the STM32CubeIDE

Good, idea. I'll try that those days

> Once we have decent tutorials/examples for using VS Code, I plan on adding VisualStudio/VGDB and STM32CubeIDE project files to the libraries, and having at least one example. But I don't have an [...]

A lot of work I guess. Feel free to ask if I can help by testing the projects on my side

stephenhensley, 2020-12-07:
Great. Hope that helps. Let me know if you still have issues once you've done that.

And I'll definitely let you know once we've got some projects ready to test!

ThomArmax, 2020-12-08:
Hi @stephenhensley I have uploaded my dummy drone "project" to my GitHub, [DaisyDrone](https://github.com/ThomArmax/DaisyDrone). I have created a simple Makefile and `build.sh` script. See README.md file.

Here are the bins resulting of the build on my side : [daisy_drone.zip](https://github.com/electro-smith/libDaisy/files/5658136/daisy_drone.zip)

stephenhensley, 2020-12-11:
Sorry for the delay on this, but I can take a look at your project tomorrow and see if there's something causing an issue either within your project or within libdaisy.

ThomArmax, 2020-12-11:
Thanks @stephenhensley 👍

ThomArmax, 2020-12-11:
Could that be the toolchain version ?
I have tried with Visual Studio 2017 and VisualGDB `5.4r11`. I'have created a solution which embed libDaisy and DaisySP projects files libs and I have exactly the same issue.

```
$ arm-none-eabi-gcc -v

Using built-in specs.
COLLECT_GCC=C:\Program Files (x86)\GNU Arm Embedded Toolchain\9 2020-q2-update\bin\arm-none-eabi-gcc.exe
COLLECT_LTO_WRAPPER=c:/program\ files\ (x86)/gnu\ arm\ embedded\ toolchain/9\ 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/lto-wrapper.exe
Target: arm-none-eabi
Configured with: /mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/src/gcc/configure --build=x86_64-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw --libexecdir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/lib --infodir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/share/doc/gcc-arm-none-eabi/html --pdfdir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-mingw-wildcard --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/arm-none-eabi --with-libiconv-prefix=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-gmp=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-mpfr=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-mpc=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-isl=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-libelf=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Arm Embedded Toolchain 9-2020-q2-update' --with-multilib-list=rmprofile,aprofile
Thread model: single
gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)

```

stephenhensley, 2020-12-11:
Okay, so I cloned your repo, and the submodules (I did update libdaisy because I just updated dfu-util and needed the latest commit for support there..).

I then ran your `build_deps.sh`, and `build.sh` before going into `src` and running `make program-dfu`

On the Daisy Pod hardware this had one knob controlling the frequency of a bunch of oscillators (slightly detuned from each other). I had the exact same result programming the binary you provided in the zip above.

The Daisy pod has KNOB 1 connected to Physical pin 28 (pin 21 in libdaisy) same as in your example's code, and the seed/Knob example.

You are using the same version of the compiler, and the resulting binary (on windows) is the same size as the one I built here.

---

If you're still having bad results with that binary file, I recommend double checking your hardware connections. Specifically that you have DGND and AGND connected, and that your pot is wired to the correct pin (matching the breadboard diagram in the Seed/Knob example). Either of those being wrong could have the effects you're describing above.

It's also worth checking continuity of your jumper wires as I have seen a fair amount of them either go bad, or be bad from the start.

Hope that helps.

ThomArmax, 2020-12-11:
Ok, I have checked my hardware, and I had a wrong cable ... :(

Thank you for your support and your patience

stephenhensley, 2020-12-11:
@ThomArmax no problem, glad you got it sorted!