daisyaudio/libDaisy #297: ADC averaging not tracking audio callback rate changes

GregBurns, 2021-01-21:
The ADC controller initialization uses the default audio frequency and blocksize to compute the slew parameters. If these are subsequently changed the slew rate is no longer valid and can be off by a lot. For example, the default audio callback frequency is 1KHz but with an audio block size of 256 at 32KHz sample rate, the callback frequency is only 125Hz. The controller Init() api is private so calling that after changing the sample rate and block size is not an option.

stephenhensley, 2021-01-25:
I can push a fix for this in the next few days.

I'll add a `SetCallbackRate` function to the `AnalogControl` class, and have the `SetBlockSize` and `SetSampleRate` functions recompute, and set this from the DaisySeed API.

stephenhensley, 2021-02-09:
So I just opened #303 which adds a SetCoeff function which I think is ultimately more useful than a 'samplerate'  or 'slewtime' control so that the filter is directly controllable.

commit 8d790dd8 (2021-05-06): Fixes #297, recompute ctrl slew on samplerate update
PR #337 (2021-05-19): Blocksize switch bugfix
Resets update rates for HIDs when board change samplerate or block size. 
Tested on all boards except version.

Small note: some HIDs refer to the update rate as 'SampleRate', and others refer to it as 'UpdateRate'.
Should probably unify that across classes. My vote goes for UpdateRate.

Fixes #317 
fixes #297
commit 845e1c01 (2021-05-19): Blocksize switch bugfix (#337)

* Add SetUpdateRate to switch and encoder

* Add setupdaterate to field switches

* Add update sample rate to ctrl.h

* Add samplerate setter for led.h

* Add field sr updates for pots and cv

* Add hid update rate update to patch

* Add petal hid update rate setters

* Add update rate HID setters to pod

* Add hid update rate setters to versio

* Fixes #297, recompute ctrl slew on samplerate update
closed 2021-05-19 by stephenhensley