daisyaudio/libDaisy #307: Mark the input of the audio callback as const

zeroisnan, 2021-02-21:
This came up while running lint checking (clang-tidy) on my code.
Any reason not to label the input of the audio callback with a const modifier?

From:
```cpp 
typedef void(* daisy::AudioHandle::AudioCallback) (float **in, float **out, size_t size)
typedef void(* daisy::AudioHandle::InterleavingAudioCallback) (float *in, float *out, size_t size)
```

To:
```cpp
typedef void(* daisy::AudioHandle::AudioCallback) (const float **in, float **out, size_t size)
typedef void(* daisy::AudioHandle::InterleavingAudioCallback) (const float *in, float *out, size_t size)
```

clang-tidy reference: [readability-non-const-parameter](https://clang.llvm.org/extra/clang-tidy/checks/readability-non-const-parameter.html)

stephenhensley, 2021-03-01:
Agreed that we should add this. I can add it to my todo. I'll be going through a lot of PRs this week. So I should be able to squeeze this in as well.

beserge, 2021-05-17:
So I've checked this out a bit. Works totally fine on libdaisy as expected.
It does require changing all of the callbacks in DaisyExamples, which is no big deal.
My one concern is that this will break a lot of user's existing code. 
It's a simple fix, but some people will be confused of course. Maybe we should put some messages out letting people know before pushing this.

PR #344 (2021-05-25): Change callback input to const float
fixes #307
closed 2021-05-25 by stephenhensley