daisyaudio/libDaisy #313: MIDI ChannelPressure message is only two bytes

GregBurns, 2021-03-13:
The Midi parser assumes that all voice messages are 3 bytes long, the ChannelPressure message is only 2 bytes long and is causing the subsequent NoteOff message to be discarded. The Aturia Keystep send the ChannelPressure message after a key is held down for a while. The fix below checks for the ChannelPressure message and changes the parser state accordingly.

 case ParserHasStatus:
            if((byte & kStatusByteMask) == 0)
            {
                incoming_message_.data[0] = byte & kDataByteMask;
                if(incoming_message_.type == ChannelPressure)
                {
                    pstate_ = ParserEmpty;
                    event_q_.Write(incoming_message_);
                }
                else
                {
                    pstate_ = ParserHasData0;
                }
            }
            else
            {
                // invalid message go back to start ;p
                pstate_ = ParserEmpty;
            }
            break;

GregBurns, 2021-04-01:
Proposed fix is in Pull Request #319

GregBurns, 2021-04-16:
Pull request has been applied.

closed 2021-04-16 by GregBurns

stephenhensley, 2021-04-16:
Oh thanks for closing this. :)

 I totally thought it was linked to the PR.