daisyaudio/libDaisy #328: Add static System function for jumping to System Bootloader

stephenhensley, 2021-04-28:
The ability to return to the System Bootloader should be added to the library. This should probably live in the System class, and also be a static function similar to the Delay functions.

On going conversations on the forum come from:

https://forum.electro-smith.com/t/boot-flash-attached-to-a-pin/804/6

and 

https://forum.electro-smith.com/t/reboot-to-bootloader-from-menu/1556/3

Tom97531, 2021-04-29:
Hi Stephen, I was about to push the code, but it looks like I cannot create a branch. I don't have the permissions.

stephenhensley, 2021-04-29:
Ah. So for contributions you'll have to create a fork of the repo, and submit a PR.

What you'll want to do is:
1. Create a fork of libdaisy (it will then be your own copy of the repo at github.com/Tom97531/libdaisy)
2. Clone your fork, create a new branch and push the desired feature.
3. Create a Pull Request from the browser with the `base` branch set to `electro-smith/master` and the `head` branch set to your feature branch.
4. Once everything is merged you can resync your master branch with the fork. Github has some instructions for this in [the docs](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/syncing-a-fork)

Let me know if you have any questions!

Tom97531, 2021-04-29:
Great I'll do that sometime today !

TheSlowGrowth, 2021-04-29:
As a hint: You can add multiple remote repos to the same local clone of the repo! 
Simply do it like this:
1. Create the fork via github.com
2. add your fork as another remote to the repo clone on your computer
3. push your existing branch to your fork
4. start the pull request directly from github.com

closed 2021-05-04 by stephenhensley