daisyaudio/libDaisy #362: Versioning and Releases

TheSlowGrowth, 2021-06-25:
Now that a first "versioned" release is coming up, I wanted to write down some thoughts about the versioning system.

My main wish is that we pick a system that follows strict and clear rules [so that we don't end up with unclear version numbers and confusion on all sides](http://sentimentalversioning.org/).
My proposal would be to use [semantic versioning](https://semver.org/) as it follows a clear logic that's suitable for a library like libDaisy. This is the gist:

> Given a version number MAJOR.MINOR.PATCH, increment the:
>    - MAJOR version when you make incompatible API changes,
>    - MINOR version when you add functionality in a backwards compatible manner, and
>    - PATCH version when you make backwards compatible bug fixes.

In many projects, the releases are completely automatic, and the [commit messages encode what type of change the commit contains](https://www.conventionalcommits.org/en/v1.0.0/). This allows the CI tools to pick the correct version number and even write changelog files automatically based on the info from the commit messages. I would love to add this to libDaisy, but it requires that all contributors follow the rules for commit messages which may discourage hobbyists and beginners from participating. Of course we can always jump in and fix the commit messages when merging a PR, but it's additional work. 

So maybe fully automatic releases are not practical yet, but nothing stops us from doing releases manually and STILL follow the semantic versioning rules.

stephenhensley, 2021-07-01:
I somehow missed this issue getting made, oops!

But I just made the first initial release as v0.1.0 since that felt like a good starting point.

I mostly agree with the above. However, I have a feeling with a number of the upcoming updates (like GPIO), and future things once we start adding various USB configurations, etc. that they major version may increment very quickly. 

That's not necessarily a bad thing, but I was mentally thinking v1.0 would be a milestone point when the entire library follows a cohesive structure, all peripherals can be used in either direction (where applicable) with blocking or DMA, and we have a composite USB device handle that can support at least CDC and MIDI simultaneously. That said, I'm not against doing anything differently.

I do love the idea of automatic releases, and I think that's something we can work toward, but I agree that we should nail down some rules and stick to them until then. 

Definitely open to discussing in more depth. I won't be merging any of the bigger changes until some time next week. So plenty of time to think it over and come up with a good plan.

TheSlowGrowth, 2021-07-02:
> I was mentally thinking v1.0 would be a milestone point when the entire library follows a cohesive structure

Yes, I understand this thought. However, realistically, a library like this will never reach a point where you are "mostly done" like e.g. a user-facing software (e.g. a game or a Eurorack module firmware). I've seen projects that had a weird versioning where they initially moved up from 0.1 to 0.5 very quickly, then realised there's still a lot to do until 1.0 so they started being much more conservative with the version increments. Ultimately, the version number looks like a "progress indicator" towards a stable 1.0 release when in reality it became a highly nonlinear thing that barely has any meaning anymore.

I think this is what semantiv versioning tries to solve: Remove the need to "map progress to version numbers" and instead focus on what the users care about: What has changed between versions? Can I safely update? Does it add new features? etc.
E.g. going from v3.4.6 to v4.0.1 tells you that you need to expect a breaking change, whereas going from v4.0.0 to v4.12.4 should go smoothly (no breaking changes!). The amount of code that changed between the two versions may have been much larger than in the former example, but this is ultimately not what a user cares about most.

recursinging, 2021-07-02:
> I have a feeling with a number of the upcoming updates (like GPIO), and future things once we start adding various USB configurations, etc. that they major version may increment very quickly.

Man I don't want to bikeshed here, but I've experienced this first hand with SemVer, and the rate at which major versions increment ends up being logarithmic.  Anyways, the numbers are almost arbitrary outside their relationship to each other.

> going from v3.4.6 to v4.0.1 tells you that you need to expect a breaking change, whereas going from v4.0.0 to v4.12.4 should go smoothly

What I've always wished for is a dynamic, structured changelog through which I can quickly parse exactly what I need to watch out for (and relevant to me) when upgrading from version X to version Y.   It looks like "conventional commits" might help to glean some machine readable structure and make something like this possible in the future.

stephenhensley, 2021-07-06:
Yup, I agree 100%. 

Having a system like this makes complete sense for the project since it really won't ever be "done". 

Thanks for the feedback, and clarification.

>What I've always wished for is a dynamic, structured changelog through which I can quickly parse exactly what I need to watch out for (and relevant to me) when upgrading from version X to version Y.

This is an interesting point, and definitely something we can shoot for in the future. 

---

Now, how would we like to implement this?

The things we need to manage are:

1. Creating tags/releases for releases
2. Ensuring the changelog, and version in CMake, and the header files are up to date
3. Handling version incrementing for parallel PRs

The first two are fairly trivial, and the tags/releases can be automated with relative ease as well.

What I'm not sure how to handle is the parallel PRs. For example, looking at the existing PRs they're most likely going to be merged in a mixed, non-chronological order since some of them aren't complete, or will have review notes to resolve.

One possible solution could be to have a "current" section of the changelog, and merge PRs to an intermediary "dev" branch. That dev branch would periodically (every week or two weeks) update the version number, and change the "current" section to the latest version, merge and create the release. This would take the burden of incrementing the version number off of the contributor, and all they would have to do is add their notes to the changelog, which could then be cleaned up, or expanded on if necessary, before the actual release.

Definitely interested in any feedback on this, or alternative approaches. Obviously don't want to tread water discussing this for days while there are perfectly good PRs to start merging 😄

TheSlowGrowth, 2021-07-06:
In an ideal world, the changelog file could entirely be generated from the commits (via the "conventional commit" formatting). This would require that we have some commit linting (=> reject commits that don't follow the formatting) which could be confusing to newcomers. Additionally each commit must contain a single change only. Both of these require some good `git` skills and I think that would scare away newcomers. 

I've had good experiences with the approach you're describing: Basically, there was a section at the top of the CHANGELOG.md file that had the headline `Unreleased`. We would accumulate relevant changes there. Whenever there was a release, the CI would replace the headline with the new version number and add a new `Unreleased` headline above it. Then the changed CHANGELOG.md file and the files containing the new version would be pushed to the repo and a tag would be placed accordingly.

On the other hand, manually editing the Changelog file will often create merge conflicts (I know that from experience). Fortunately they're relatively easy to solve, but it's still an additional step. We will often end up in situations where there are two PRs, both green and ready to go. After merging one, the other will be red due to a CHANGELOG.md merge conflict and need further attention from the author. This will slow down the process of merging a little bit and it has the potential to be frustrating for the PR authors. 

MAYBE there's another option: What if we could have a CI job that does the merging and adds the CHANGELOG.md entry in the process (and does it in a way that will avoid merge conflicts)? There are workflow triggers we could use:
- [`workflow_dispatch`](https://docs.github.com/en/actions/reference/events-that-trigger-workflows#workflow_dispatch) trigger could be used to manually trigger the merging while at the same time providing the changelog.md content as parameters. 
- [`pull_request_review_comment`](https://docs.github.com/en/actions/reference/events-that-trigger-workflows#pull_request_review_comment) trigger could merge the PR automatically when triggered by a special comment in the PR discussion. 
Such a comment could look like this:
```
MERGEBOT
- fix: fixed crash when sending certain combinations of MIDI messages
- feat: added DMA support to SpiHandle
- BREAKING CHANGE: changed GPIO to a modern C++ API
```
Accessing the comments and merging the PR from a CI job can be done with the REST API: https://docs.github.com/en/rest/reference/pulls#comments
Since repo maintainers would trigger the merging, they could edit the changelog entries according to a standardized format so that the CI job can add a CHANGELOG.md update commit immediatels before merging.

stephenhensley, 2021-07-07:
Good points. I hadn't thought about the merge conflicts needing to be resolved on the PR authors end. That could be a bit frustrating.

I do like the idea of the CI process idea, using a special PR comment prior to merge could be a good solution to that issue.

Until we get that set up (I won't have time to set up any of the release CI stuff for another week or two), we can use the manual method.

So I guess last question is do we need/want a `dev` branch, or do we still want to work from master, and just have tagged releases from that branch? I like the idea of master being the latest tagged release for the sake of new people cloning the repo, but it's also not critical since integrations can leave the submodules on the desired release themselves.

TheSlowGrowth, 2021-07-07:
The problem with github repos is that the "main" branch is the one that people will clone by default - and it's also the one that PRs go to by default. With a `master`/`dev` approach, you'd either have `dev` as the main branch, destroying the "initial checkout == stable" benefit, OR you'd have `master` as the main branch, but then all PRs will default to target the `master` branch, which is wrong.

I guess for simplicity I would vote for a single `master` branch.

stephenhensley, 2021-07-09:
Yeah agreed. 

We'll use master to keep it simple.

Also, the changelog merge conflicts are definitely going to be a bit annoying.  I'll try to carve out some time in the next week or two to get the automated with the PR comments as discussed.

Thanks everyone for the feedback, and discussion!

closed 2021-07-09 by stephenhensley