daisyaudio/libDaisy #363: Report memory usage at the end of the build step

zeroisnan, 2021-06-26:
**How to**:
Append the following to the `LDFLAGS` variable in `core/Makefile`:
```
 -Wl,--print-memory-usage
```

I am not a CMake expert, but I assume you should append the same to `cmake/default_build.cmake` in the section `target_link_options(${FIRMWARE_NAME} PUBLIC`

**Example output**:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       32440 B       128 KB     24.75%
         DTCMRAM:          0 GB       128 KB      0.00%
            SRAM:      271844 B       512 KB     51.85%
          RAM_D2:         16 KB       288 KB      5.56%
          RAM_D3:          0 GB        64 KB      0.00%
         ITCMRAM:          0 GB        64 KB      0.00%
           SDRAM:          0 GB        64 MB      0.00%
       QSPIFLASH:          0 GB         8 MB      0.00%
```

Source: https://embeddedartistry.com/blog/2020/08/17/three-gcc-flags-for-analyzing-memory-usage/

I can prepare a PR if this change is of interest

TheSlowGrowth, 2021-06-27:
Please make a PR! I'll be using CMake in the future and would find this very useful

stephenhensley, 2021-07-01:
Agreed! This would be great!

PR #379 (2021-07-19): Report memory usage at the end of the build step (fix #363)
closes #363 - added linker options as described in the ticket
commit 6946b375 (2021-07-19): Report memory usage at the end of the build step (fix #363) (#379)

* Report memory usage at the end of the build step (fix #363)

* remove now redundant size invocation; updated CHANGELOG
closed 2021-07-19 by stephenhensley