daisyaudio/libDaisy #471: MIDI 2.0

2fd5, 2022-01-06:
Is MIDI 2.0 in the pipeline ?

I would love to see it in libDaisy

As MIDI 2.0 is the future and I think of devkits as a way to prototype devices for the future it make sense to experiment with 2.0

I understand that these are early days but we need to move that chicken and egg issue 

Thanks for considering

stephenhensley, 2022-01-07:
@2fd5 MIDI 2.0 is definitely in the pipeline, but we do not currently have a timeline.

We are very excited to work on adding it to libDaisy. However, we are still working through a number of the core features of the library that have a higher priority.

I will leave this issue open as a place to discuss implementation if someone else would like to start working on it, and will update here as well when our team begins work on it.

If you know of any particular gear that leverages MIDI 2.0 well, I'd be interested to know what people are interested in, and using already!

thealienthing, 2022-10-26:
I'm using the daisy platform for my masters project and have considered MIDI 2.0 as a potential avenue to explore. Does anybody know of any inexpensive off the shelf devices that supports MIDI 2.0 protocol? If it does, I would definitely be interested in helping add MIDI 2.0 support to libDaisy.