daisyaudio/libDaisy #486: [MIDI] Basic output generation w/out dealing with bytes directly

stephenhensley, 2022-02-21:
This was originally intended to be added when we added parsing, and it must have just gotten glossed over.

There are some ideas in [this forum post](https://forum.electro-smith.com/t/how-to-use-libdaisy-for-midi-out/2346/6).

Also, as an alternative, we could flesh out the `MidiEvent` types a bit more, and have `SendMessage` dispatch those.

Overall, open to ideas, but currently the only method to deal with MIDI out is to directly manage the bytes.

makingsoundmachines, 2022-03-07:
A great source of inspiration is the MIDI lib used by Teensy:

https://github.com/FortySevenEffects/arduino_midi_library

I often look here when I need a specific snippet of code for MIDI output.

SnazzyBlanket, 2022-07-15:
any update on this?

stephenhensley, 2023-01-26:
@SnazzyBlanket thanks for the bump here, I must have missed the notifications.

We're wrapping up some improvements to the transport in #557, and then I think it'll be pretty easy to add some message specific functions like the teensy library above.

Something like:

```cpp
midi.SendNoteOn(chn, nn, vel);
midi.SendNoteOff(chn, nn, vel);
midi.SendControlChange(chn, cc, val);
```
etc. 

I'll try to shoot for having this in there by end of February :)

SnazzyBlanket, 2023-01-26:
That would be great 

Thanks for the heads up

filipforsstrom, 2024-01-18:
Any update on this? :)

willemOH, 2024-05-30:
@stephenhensley Any updates on this? Wanting to know if there's any support from libDaisy before I go hacking it together myself. If anyone else has done something like this please let me know