daisyaudio/libDaisy #520: [Question] Possible to use JUCE

jakemumu, 2022-05-02:
Hey! I see Max and some other audio frameworks listed as possible to be used on the Daisy -- I'm curious if it's possible to run JUCE apps on the Daisy. 

Thanks! 

Jacob

TheSlowGrowth, 2022-05-03:
JUCE is a framework for C++ code. Apart from utility classes for all sorts of things like Strings/Graphics/GUI, it provides low level abstractions so that you can run the same code on Win/Mac/Unix/Android/iOS, as VST/AU/AAX plugins, etc. I would say at least 98% of JUCE is completely irrelevant to hardware like Daisy. Consider it more like a "virtual machine" or an "operating system" that allows code to work independently of the platform.

For a hardware based on Daisy, there is no need for VST plugins, or operating system abstractions. 

However, you can of course write C++ code that runs as part of a JUCE app and as part of Daisy-based hardware.

jakemumu, 2022-05-03:
Thanks yes -- this I'm aware of however JUCE has a powerful DSP framework and graph system. I have a lot of apps built in JUCE which I'd like to port their processing functions to run on Daisy, but removing JUCE as a dependency would be a large issue as the DSP code has JUCE's AudioBuffer & other DSP classes as dependency.

TheSlowGrowth, 2022-05-03:
You could try to manually include various JUCE headers and build your own makefile / CMake setup to target the embedded hardware platform. It's not going to be easy, not because Daisy can't run the code in the end, but because JUCE is not exactly easy to integrate in any non-JUCE environment. Even if all you wanted is the JUCE graph processing, the JUCE module system still requires you to add most of the core JUCE modules which already contain OS abstractions and other code that definitely won't run/compile for Daisy.
I suspect it can be done, but it would be a lot of work and you'd need to heavily edit/patch JUCE code to throw out all the OS sepcific stuff.

On top of that, JUCE is not really designed to be lightweight. It relies a lot on dynamic memory allocation (which is troublesome on embedded hardware like Daisy) and unless you patch out most of the utilities like String classes, you'd end up with a Daisy firmware that by far exceeds the 128kB internal flash memory of the Daisy.

jakemumu, 2022-05-03:
Got it! That clears it up thanks @TheSlowGrowth -- to be honest I wasn't sure what the firmware here was and thought it may be a linux variant that JUCE could compile for. Appreciate your response!

closed 2022-05-03 by jakemumu