daisyaudio/libDaisy #521: Running not handled for single byte commands

hammondeggs, 2022-05-05:
https://github.com/electro-smith/libDaisy/blob/04dd33614ab67fde6deae3ec0bc60a0ad128e3c3/src/hid/midi.h#L217

**Issue:**
Currently here we do not handle running status for single-byte commands (Program Change, MTC Quarter Frame,Song Select, and really, more importantly, _Channel Pressure_). This appears to cause dropped channel pressure messages resulting in a "stuck" pressure status.

**Method of reproduction:**
Using a Yamaha Motif XF, play and hold a chord, and quickly press more firmly and release while still holding the chord to quickly engage/release the channel pressure repeatedly. The synthesizer appears to occasionally send running status messages for channel pressure. If at the very least you track the 'latest' channel pressure message, this should always reset to 0 once all of the keys are released, however currently it may not, as single byte running status messages are being dropped.

**Suggested fix:**
I appear to have fixed this by updating this code to:
```
incoming_message_.type    = running_status_;
incoming_message_.data[0] = byte & kDataByteMask;
//check for single byte running status, really this only applies to channel pressure though
if(running_status_ == ChannelPressure
   || running_status_ == ProgramChange
   || incoming_message_.sc_type == MTCQuarterFrame
   || incoming_message_.sc_type == SongSelect)
{
    //Send the single byte update 
    pstate_ = ParserEmpty;
    event_q_.Write(incoming_message_);                        
}
else
{
    pstate_                   = ParserHasData0; //we need to get the 2nd byte yet.
}

```

stephenhensley, 2022-05-18:
Thanks for the issue report!

We'll get on fixing this ASAP.

If you have some hex output like you did in #522 that would be helpful for us to write a test against to make sure the fix works!

Thanks!

stephenhensley, 2022-05-19:
@hammondeggs thanks again for the fantastic issue report; should be fixed on the latest master branch.

Let us know if you catch any other issues!

PR #525 (2022-05-19): Add bugfix for single data byte running status, add unit test
fixes #521
closed 2022-05-19 by stephenhensley