daisyaudio/libDaisy #535: WavPlayer digital clicks/playback issues

stephenhensley, 2022-07-11:
Reported on the [forum](https://forum.electro-smith.com/t/periodic-digital-clicks-when-reading-file-from-sd-card-on-daisy-seed/2750/); seems like something to do with `Prepare` is causing audio to be held up, and interferes with the callback in general. 

This was demonstrated by generating a low freq. oscillator tone while running the `WavPlayer` and hearing some digital artifacts.

netherwaves, 2022-07-14:
@stephenhensley hello! reporting back on this:

i've seemed to fix the problem by setting the buffer array as a global object with the `DSY_SDRAM_BSS` attribute like so:
```
#define BUFSIZE 4096
int16_t DSY_SDRAM_BSS buffer[BUFSIZE];
```
and by modifying the `Init`, `Prepare` and `Stream` methods of `WavPlayer` to be able to receive this array as an argument and thus modify it in real time. the clicks are now gone! yippie!

it doesn't seem like the attribute can be added directly on the class property buffer `buff_` so i will simply use this method from now on locally :)

would it be a good idea to start a pull request to address this bug with my fix? i could also change the `DaisyExamples` entry to reflect the changes if we go forward with this

stephenhensley, 2022-07-18:
Hey @netherwaves this is a really interesting update. Thanks for sharing

When the buffer was internal to the WavPlayer, what memory section did you have it in? 
Was it just declared globally?

I'm generally a fan of portable buffers within libDaisy to changing sizes, and using other sections without having to have the object's logic data in the other memory. So I'd be open to a PR to change this.

However, it does make me wonder if there's some issue with the classes use of that buffer possibly overflowing beyond the size of the buffer or something.
In which case, the portable buffer would just make it a bit more difficult to notice it.

netherwaves, 2022-07-20:
@stephenhensley 
> When the buffer was internal to the WavPlayer, what memory section did you have it in?
> Was it just declared globally?

ah! so you mean it would be an idea to try something like this then:
```
WavPlayer DSY_SDRAM_BSS sampler;
```

this isn't something i had thought of, and yes you're right, it's been declared globally all this time, since i've been using the DaisyExamples entry as a boilerplate. although i just tried it and it doesn't fix the problem... if anything, the clicks are now even louder than before :p dismembering the buffer itself is still what seems to work best for me

captainkirkby, 2022-10-12:
+1 to this issue, when I use the example .wav playback with no modifications, and an SD card with a full scale I see and hear glitches in the waveform.
[image]

GR8Instruments, 2022-11-02:
Hi, I am wondering, is that the wave header accidently getting copied into the audio stream?

tanilas, 2022-12-17:
Same trouble here, I tried the workarounds mentioned above, also putting WavPlayer into the SDRAM (DSY_SDRAM_BSS). No luck this far.

VorkHeftruck, 2023-03-08:
+1. Is this issue being addressed to, it seems awfully quiet here?

VorkHeftruck, 2023-03-08:
@netherwaves would you mind sharing your code on his? I don't fully understand now you passed/used the array in sdramm?

netherwaves, 2023-03-18:
@VorkHeftruck here's a gist with a modded version of the WavPlayer class (located at `libDaisy/src/hid/wavplayer`), accompanied by an usage example script: https://gist.github.com/netherwaves/880785ccac5ddf5907ba187c16241ecc

VorkHeftruck, 2023-03-18:
@netherwaves thanks for the code, very insightful! I've tried it locally (I changed it a bit to work outside of the library). I understand what you meant in previous messages, now. You pass the SDRAM buffer as a pointer to the Init. The other changes you made are very similar to my own. However the clicks and digital artifacts are still there... Way less. But still...
But thanks for your example anyway, much appreciated!