daisyaudio/libDaisy #541: cannot f_open from SD card when using SRAM bootloader

netherwaves, 2022-08-25:
i've recently had to implement the SRAM bootloader on a Daisy Seed for a project to compensate for my program's large size, and need to access an SD card for a JSON registry and .wav audio files. however, the program returns an error 1 (FR_DISK_ERR) after around ten seconds of stalling on a call to `f_open`. `f_mount` on the root directory returns FR_OK.

here is the code i have tried running on the internal flash as-is, on the SRAM memory with the bootloader and the QSPI memory. needless to say, it works normally on the internal flash; it doesn't on SRAM and QSPI

```
#include "daisy_seed.h"

using namespace daisy;

DaisySeed hw;
SdmmcHandler sdcard;
FatFSInterface fsi;

FIL file;

int main(void)
{
    hw.Init();
    hw.StartLog(true);

    System::Delay(1000);

    SdmmcHandler::Config sd_config;
    sd_config.Defaults();
    sdcard.Init(sd_config);
    fsi.Init(FatFSInterface::Config::MEDIA_SD);
    f_mount(&fsi.GetSDFileSystem(), fsi.GetSDPath(), 0);

    System::Delay(1000);

    hw.PrintLine("opening...");
    if(int result = f_open(&file, "registry.json", (FA_OPEN_EXISTING | FA_READ)))
    {
        hw.PrintLine("COULD NOT OPEN - %ld", result);
    }
    else
    {
        hw.PrintLine("OPENED OK");
    }
}
```

and here is my Makefile
```
# Project Name
TARGET = Sandbox
APP_TYPE = BOOT_SRAM
USE_FATFS = 1

# Sources
CPP_SOURCES = main.cpp

# Library Locations
LIBDAISY_DIR = ../libDaisy
DAISYSP_DIR = ../DaisySP

# Core location, and generic makefile.
SYSTEM_FILES_DIR = $(LIBDAISY_DIR)/core
include $(SYSTEM_FILES_DIR)/makefile
```

i am suspecting the bootloader of monopolizing the SD card management by virtue of its scan search for a .bin file on the device when it is connected. i absolutely do not need the SD card for that purpose, so this is a little annoying considering the internal flash seems to be my only option now.

is there something 
thanks!

stephenhensley, 2022-08-25:
So, this is more likely to do with the default layout of the SRAM linker script, which puts the default `.bss` (uninitialized memory) into the DTCMRAM on the STM32.

That memory is unable to communicate with the SDMMC peripheral. 
The `FIL` object has to be in a location that can use that peripheral.

There are a few solutions:

1. You can use a linker script that reduces the "program space", and still utilizes the AXI SRAM for bss (albeit less), and leave your code as-is. 

An example of the linker a linker script is [here](https://github.com/electro-smith/DaisyExamples/blob/master/field/Nimbus/nimbus.lds). The `SRAM_EXEC` section size determines the "program memory", while the `SRAM` section is your default internal bss size.

2. You can move the `FIL` object above into the external SDRAM (the only other memory section capable of communicating with the SDMMC).

This can be done by adding an attribute to the global variable:

```cpp
FIL DSY_SDRAM_BSS file;
```

Hope that helps!

netherwaves, 2022-08-25:
@stephenhensley hi! thank you for the quick response.

here is a list of the things i have tried since your comment:
- using the default SRAM script while FIL is on .sdram_bss (DSY_SDRAM_BSS attribute);
- using the nimbus linker script while FIL is on .bss (no attribute);
- using the nimbus linker script while FIL is on .sdram_bss;
- changing the default SRAM script so .bss goes to `> DTCRAM AT > SRAM`

nothing here seems to have worked, unfortunately... i am getting a consistent error log in each of those scenarios
```
f_mount 0
opening...
COULD NOT OPEN - 1
```

is there something i'm still missing here? linker scripts were completely new to me up until today, so i might not fully understand what is going on here...

netherwaves, 2022-08-25:
okay! my bad, it seems like loading the program on the QSPI does allow for SD card operations. i must've messed up somewhere while testing it but `APP_TYPE = BOOT_QSPI` returns FR_OK on f_open.

this is still very strange to me though, from looking at the QSPI linker script it doesn't seem there is anything different from what i've been trying to tweak in the SRAM linker, aside from `.text` being loaded on the QSPI flash rather than the SRAM. do you have any ideas as to why this may be happening?

stephenhensley, 2022-08-25:
hmm, glad to hear you got it functioning on `BOOT_QSPI`. 

I could have sworn I've done some SDMMC stuff with the `BOOT_SRAM` option. I've definitely done a few things with FatFS and USB drives while using the bootloader.

There shouldn't be much of a difference between the SRAM/QSPI linkers except that nearly all of the AXI SRAM is used for program space in the default SRAM linker.

In the case of using the `.sdram_bss` section, it may also be required to put the `FatFSInteface` in that section as well.

---

What return code do you get from your `f_open` call?

netherwaves, 2022-08-26:
> What return code do you get from your `f_open` call?

so far it's always been 1 (FR_DISK_ERR)

> In the case of using the .sdram_bss section, it may also be required to put the FatFSInteface in that section as well.

i suspected that too, i think i tried it but it didn't change anything... i should definitely try it again though! it's entirely possible that this entire situation is caused by distraction errors, so i'll give it another proper go and report back if i can get a consistent error behaviour outlined. in any case i can confirm without a doubt that the QSPI linker script works :)

netherwaves, 2022-11-07:
hi there! reporting back on this issue, i've come back to the linker script business and have found that the nimbus.lds file you've mentioned earlier works in conjunction with putting the `DSY_SDRAM_BSS` attribute on the `FIL` object! so i would consider this issue closed for now, as i can now manage to run my program with all features from the SRAM. thank you, and sorry for the inconvenience!

closed 2022-11-07 by netherwaves