daisyaudio/libDaisy #588: Logger <LOGGER_INTERNAL> doesn't work (maybe concurrency issue)

ohmtech-rdi, 2023-07-25:
### Description

When using the Logger over "internal USB", logging either does work, or doesn't work at all, or just a few frames of logs are received and then it doesn't work anymore.

- This is dependent on the program
- Sometimes it works, sometimes not
- Sometimes launching the program using openocd makes it work, or the inverse

### Analysis

```
template <LoggerDestination dest>
void Logger<dest>::StartLog(bool wait_for_pc)
{
    impl_.Init();   // will trigger OTG_FS_IRQHandler interruption at some point

    /* if waiting for PC, use blocking transmission */
    pc_sync_ = wait_for_pc ? LOGGER_SYNC_IN : LOGGER_SYNC_OUT;
    /** transmit something to stall the UART until a terminal is connected
     * at least two separate calls are required
     */
    PrintLine("Daisy is online");
    PrintLine("===============");
    System::Delay(10);
}
```

After `impl_.Init();` is called, `PrintLine` will immediately start to `CDC_Transmit_FS`.
At some point of time, a `OTG_FS_IRQHandler` interruption will be fired, in which it will setup the CDC.
Typically when the bug happens:

```
CDC_Init_FS@0x08008e2e (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/usbd/usbd_cdc_if.c:178)
USBD_CDC_Init@0x080136bc (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c:546)
USBD_SetClassConfig@0x0801382e (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c:231)
USBD_SetConfig@0x08013cf0 (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c:584)
USBD_StdDevReq@0x08013e40 (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c:136)
USBD_LL_SetupStage@0x0801388a (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c:272)
HAL_PCD_SetupStageCallback@0x08009022 (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/usbd/usbd_conf.c:212)
PCD_EP_OutSetupPacket_int@0x0800e804 (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_pcd.c:2188)
HAL_PCD_IRQHandler@0x0800e9da (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_pcd.c:1083)
OTG_FS_IRQHandler@0x080067b4 (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/hid/usb.cpp:173)
<signal handler called>@0xffffffe9 (Unknown Source:0)
daisy::UsbHandle::TransmitInternal@0x08006786 (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/hid/usb.cpp:128)
daisy::LoggerImpl<(daisy::LoggerDestination)1>::Transmit@0x0800657e (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/hid/logger_impl.h:61)
daisy::Logger<(daisy::LoggerDestination)1>::TransmitSync@0x0800657e (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/hid/logger.h:113)
daisy::Logger<(daisy::LoggerDestination)1>::TransmitBuf@0x0800657e (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/hid/logger.cpp:76)
daisy::Logger<(daisy::LoggerDestination)1>::PrintLineV@0x08006630 (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/hid/logger.cpp:44)
daisy::Logger<(daisy::LoggerDestination)1>::PrintLine@0x0800664e (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/hid/logger.cpp:32)
daisy::Logger<(daisy::LoggerDestination)1>::StartLog@0x08006672 (/Users/raf/Desktop/dev/eurorack-blocks/submodules/libDaisy/src/hid/logger.cpp:56)
main@0x0800189c (/Users/raf/Desktop/dev/eurorack/frohmage-erb/artifacts/perf/main.cpp:154)
```

The first time `CDC_Transmit_FS` is called:

```
uint8_t CDC_Transmit_FS(uint8_t* Buf, uint16_t Len)
{
    uint8_t result = USBD_OK;
    /* USER CODE BEGIN 7 */
    USBD_CDC_HandleTypeDef* hcdc
        = (USBD_CDC_HandleTypeDef*)hUsbDeviceFS.pClassData;
    if(hcdc->TxState != 0)
    {
        return USBD_BUSY;
    }
    USBD_CDC_SetTxBuffer(&hUsbDeviceFS, Buf, Len);
    result = USBD_CDC_TransmitPacket(&hUsbDeviceFS);
    /* USER CODE END 7 */
    return result;
}
```

`hcdc` is `NULL`, and it seems that `hcdc` is big enough for the `TxState` to be mapped to an address that can be at least read.

### Workaround 

I introduced a delay between `impl_.Init();` and before `CDC_Transmit_FS` is first called:

```
template <LoggerDestination dest>
void Logger<dest>::StartLog(bool wait_for_pc)
{
    impl_.Init();   // will trigger OTG_FS_IRQHandler interruption at some point

    System::Delay(1000); // wait for CDC_Init_FS to be called

    /* if waiting for PC, use blocking transmission */
    pc_sync_ = wait_for_pc ? LOGGER_SYNC_IN : LOGGER_SYNC_OUT;
    /** transmit something to stall the UART until a terminal is connected
     * at least two separate calls are required
     */
    PrintLine("Daisy is online");
    PrintLine("===============");
    System::Delay(10);
}
```

And that seems to solve it, at least when the program is living in the `FLASH` section.

takumi-ogata, 2023-08-04:
Hi Raphael! I'm sorry for the delay in response. 
Thank you for documenting and reporting the issue as well as a temporary workaround.

We'll have a closer look at this when we get a chance.

lerouxb, 2024-12-16:
I seem to be getting the same issue. Serial works very intermittently. Even just whittling down the example to the basics the serial port doesn't show up on mac anymore:

```
#include "daisy_seed.h"

using namespace daisy;

DaisySeed hw;

int main(void)
{
    hw.Init();
    hw.StartLog();
  
    while (1) {
        hw.PrintLine("hello");
    }
}
```

Tried different cables (even though these all worked in the past), tried adding the `System::Delay(1000);` as above, tried it in various places.

Weirdly this worked at times in the past. Even this minimal example.

darrengit, 2026-08-25:
I chased what looks like this exact family of symptoms (CDC console that “sometimes works, sometimes carries a few frames, then nothing”) on a Daisy Patch and found two independent causes worth checking here:

The suspend callback gates the USB PHY clock and nothing ever ungates it (filed as #716 with register-level evidence). One transient bus suspend — e.g. the host briefly suspending the port around driver attach — and the port is dead both ways until reset, while still enumerated. This matches “works, or doesn’t work at all, or a few frames and then nothing”, and the macOS reports downthread: macOS suspends idle CDC ports aggressively.

The detail in the original report that launching under openocd changes the behavior is itself a clue: an attached debug probe sets the DBGMCU sleep/debug-clock bits, which changes low-power behavior. On my board, application code sleeping in plain WFI (SLEEPDEEP=0) could fail to ever wake — reproducibly fatal without a probe attached and invisible with one. If the logging application (or anything in its loop) executes WFI, that interaction is worth ruling out: our single-variable test was one added wfi in an otherwise-working firmware’s idle loop, which took a 20/20 console test to 0/20.

Also worth noting for heavy use: CDC_Receive_FS re-arms the OUT endpoint unconditionally, so sustained input faster than the application drains is silently dropped; re-arming only when there is buffer room (letting the endpoint NAK) gives lossless flow control.