daisyaudio/libDaisy #595: Color::Init() initialization bug

Anstapen, 2023-11-17:
There is a (easy to fix) bug in the Color::Init(float red, float green, float blue) function, probably introduced by copying source code.
In Line 40 of src/util/color.cpp the member variable gets assigned a wrong function parameter:

`green_ = clamp(red, 0.0f, 1.0f);`
should be:
`green_ = clamp(green, 0.0f, 1.0f);`

takumi-ogata, 2023-11-21:
Hi Anstapen, 

Good catch! If you want to, you're more than welcome to make that change and submit a PR.
We would greatly appreciate it. Otherwise, we will fix the issue when we get a chance.

Thank you so much for bringing this to our attention.

Anstapen, 2023-11-21:
Hi,
sure, will do that!

Best Regards,
Anton