daisyaudio/libDaisy #602: Feature: Trill sensor support

dromer, 2023-12-18:
I was mentioned on the forum some ~2 years ago, but here an official ticket to request and track this.

We can already use Trill sensors using the Arduino library, but it would be great to also have them in libDaisy as a lot of projects use this rather than Arduino.

takumi-ogata, 2023-12-20:
Thank you for the request! I personally would love to see this happen too. Using Trill with DaisyDuino is a lot of fun, so it would be great to have the option to use it with other languages. I believe expanding the support will lead to many amazing projects!

dromer, 2024-02-23:
@giuliomoro created a PoC for this: https://forum.electro-smith.com/t/daisy-x-bela-trill-getting-started-guide/835/7

However I have not been able to make it work yet.