daisyaudio/libDaisy #608: I2C_4 seemingly broken

jgoney, 2024-01-30:
I posted on the forum about this [here](https://forum.electro-smith.com/t/cant-use-i2c-4-on-seed/4610), but tl;dr is that I can't get I2C_4 to work at all. I have no problems getting I2C_1 to work (on either set of supported pins), but I2C_4 does not work for me on either pin set. 

I made a Gist for it [here](https://gist.github.com/jgoney/2a554e1f80a943f9936f04a7061d03d2). If anyone could try it out just to sanity check it for me, that would already be a huge help.

Please let me know if you need any additional info or context.

AndrewCapon, 2024-05-04:
It isn't working here either if it makes you feel any better :)

AndrewCapon, 2024-05-04:
Ok, I think I have fixed it.

line 634 of i2c.cpp in libDaisy should be:
```
GPIO_InitStruct.Alternate = GPIO_AF6_I2C4;
```

not
```
GPIO_InitStruct.Alternate = GPIO_AF4_I2C4;
```

AndrewCapon, 2024-05-04:
https://github.com/electro-smith/libDaisy/pull/628

jgoney, 2024-05-04:
@AndrewCapon thanks for taking a look at this! I'll test your your fix as well on Monday. 👍🏻

jgoney, 2025-03-03:
This is fixed by #628, closing as fixed.

closed 2025-03-03 by jgoney