daisyaudio/libDaisy #631: Buggy shift register input on Field using arm-gcc 12.2

dromer, 2024-06-14:
Yesterday I was experimenting with the latest 7.0.1 release and when using my system arm-gcc (`12.2.1`) the pad inputs with the shift register (`CD4021`) behave very erratic.

Pressing a single button results in many additional button presses and other buttons being triggered.
Reverting back to `10.2.1` the behavior is stable again.

More close inspection to what causes this glitching (and if other parts of the stack are affected) would be in order.

It would be great if we could move to newer GCC as having to rely on such an old version will get more annoying down the line.

stephenhensley, 2024-10-08:
I recently experienced this without updating the compiler, but by updating an entire project to the latest libDaisy (it was a fairly old version of libDaisy that it started on, but the HAL drivers/GPIO refactor felt like the most likely to have caused slight timing issues.

I have since added the ability to externally set the amount of delay in ticks on the SR 4021 shift register, and increased the default. It seems like timing was barely on the edge of what was acceptable for the 4021 running at 3.3V (the datasheet only documents behavior down to 5V). 
I may further increase the default over the next week or so as I have found a few other scenarios where the buttons work, but extraneous noise in the system could cause spurious button presses on things connected. The new setting I'm testing with seems extremely stable. I might be able to find some time to test this version on the latest compiler in the next week or so.

dromer, 2024-10-09:
Oh interesting! I hope to get back to testing newer libDaisy+bootloader soon-ish and will try different compiler versions with that as well.

Should've saved an example patch/program for this ticket so I could reproduce.