daisyaudio/libDaisy #634: SSD1306 OLED display coordinates are offset

Len42, 2024-09-02:
I'm using a 128x32 pixel SSD1306 SPI OLED display connected to a Daisy Seed, using daisy::OledDisplay&lt;daisy::SSD130x4WireSpi128x32Driver&gt;. Whenever I display anything, it is offset 32 pixels in the X direction. For example, this code:
```
display.Fill(false);
display.SetCursor(0,0);
display.WriteString("abcdefghijklmnop", true);
display.Update();
```
displays this:
[image]

The letter "a" should appear at (0,0) but it shows up near the middle of the alphabet.

I tracked the problem to this line of code in libDaisy:
https://github.com/electro-smith/libDaisy/blob/f7727edb9a1febdd174b5310a7bc65340dae8700/src/dev/oled_ssd130x.h#L379

If I delete that line, the display works fine:
[image]

But it looks like that line of code was specifically added for 32-pixel-high displays, so I suspect that simply deleting it isn't the correct solution. Maybe different SSD1306 modules aren't perfectly compatible?

stephenhensley, 2024-10-08:
I was working with a similar display, and had this same issue, and was also able to resolve it by commenting out that line.. 
At the time, I thought I was the unique one. Might just be the original author for 32-pixel-height had a display that was slightly different.

Might be worth seeing if/which displays require that extra high-column-addr offset, and making a separate driver for it, with most/everything else shared with the exitsting 130x driver.

dredzik, 2025-08-13:
Same issue here. 

[image]