daisyaudio/libDaisy #664: Missing drivers

Richfield-Brucellosis, 2025-04-03:
It appears that the current release is missing drivers. Specifically, the STM32H7xx_HAL_Driver, CMSIS-DSP, STM32H7xx, and CMSIS_5 folders are empty.

dromer, 2025-04-03:
You will need to pull submodules.

 ie. like this: `git submodule update --init --recursive`

Richfield-Brucellosis, 2025-04-03:
How would I do this with the release download? It has some differences with the source code (in the cmake folder) so I'm not using git clone. (sorry if this is a dumb question I don't use github very often).

dromer, 2025-04-03:
It's better to use git clone. The release downloads indeed do not contain submodules and shouldn't really be used in this case.

stephenhensley, 2026-10-20:
The releases of libDaisy do not include the git submodules (repositories from ARM, ST, etc. containing required drivers), and they cannot be added from the release files without using git.

We do still create a fully cloned version of the project, libraries, and these dependencies when doing releases of the [DaisyBlinkProject](https://github.com/daisyaudio/DaisyBlinkProject).
So if using `git` isn't something you'd like to do, you can download the compiled .zip file from the [latest release](https://github.com/daisyaudio/DaisyBlinkProject/releases/latest) of that repo and that will have all necessary files.

However, in general, we recommend to use `git` when working with the libraries.
You can clone the repo with all of its dependencies by using:

```
git clone https://github.com/daisyaudio/libDaisy --recurse-submodules
```

or use the command @dromer mentioned above after cloning without the `--recurse-submodules` flag.

closed 2026-10-20 by stephenhensley