daisyaudio/libDaisy #666: Midi In Clock not supported

willemOH, 2025-04-17:
Currently MidiEvent.h states that only 3-byte messages are handled properly at this time. Clock messages are 1-byte. 

I experienced how improperly they are handled when I tried to figure out for a day why my incoming midi notes channels were changing inexplicably. It was my device sending midi clock which corrupted the stream. 

Fix: support Clock In

dromer, 2025-04-17:
I have used midirealtime messages in the Heavy Compiler wrapper for Daisy: https://github.com/Wasted-Audio/hvcc/blob/develop/hvcc/generators/c2daisy/templates/HeavyDaisy.cpp#L117-L145

willemOH, 2025-10-04:
@dromer did you use this wrapper in combination with libDaisy/C++ or was the project purely with Pure Data? Still looking for a good solution or fix here for a project made with libDaisy.

dromer, 2025-10-04:
@willemOH this uses libDaisy and C/C++, there is no Pure Data involved.

closed 2026-04-01 by stephenhensley